esphome/esp-web-tools #16: [Feature] Disable the button if the linked manifest or files in the manifest are not available

TD-er, 2021-06-15:
Would be nice to have the generated button be disabled if the linked file is not (yet) available.
And maybe switch color if only ESP8266 or ESP32 is present. (or add some marker in the button, whatever is more esthetical pleasing)

balloob, 2021-06-15:
I think that both of these requests shouldn't be handled by esp-web-tools but instead be handled by the website that hosts the button.

I do think that we could incorporate a `disabled` feature to disable the built-in button.

TD-er, 2021-06-15:
My idea to ask it here was that the manifest parsing and interpreting is part of this code.
But it is sure debateble whether it is the responsibility of this code, or the calling code to signal whether a flash can be completed or not.

balloob, 2021-06-15:
I understand. 

Since the manifest attribute is set by the website, the website should be able to know if it ESP8266/ESP32 and act accordingly (like passing in a button which shows such info into the button slot). We keep our button as minimalist and instead allow swapping it out completely if you want to customize it.

TD-er, 2021-06-15:
OK.
This does require some "active" code outside this button to detect the presence of a file.
So I cannot use static HTML to create an overview page.
This will make my build script a bit more elaborate... and even worse, will show my lack of JS knowledge.
Let's hope Stack Overflow isn't offline the next few hours, or else I loose my JS programming skills completely. ;)

balloob, 2021-11-11:
This is not something we'll implement in ESP Web Tools.

closed 2021-11-11 by balloob