esphome/esp-web-tools #21: install and then?

tcurdt, 2021-06-18:
I tried both

https://www.home-assistant.io/blog/2021/06/16/power-up-your-esp-projects/
https://esphome.github.io/esp-web-tools/

it installs a generic esp8266 (although it is a NodeMCU) just fine - but then nothing happens.

When I press "Connect to Wifi"

https://www.home-assistant.io/blog/2021/06/16/power-up-your-esp-projects/

it scans - but there are no devices to pair.

balloob, 2021-06-18:
ESP8266 devices do not have bluetooth and so are not able to use Improv. You will need to find their Wi-Fi access point and configure their Wi-Fi like that. We're working on making that process smoother.

closed 2021-06-18 by balloob

tcurdt, 2021-06-18:
@balloob makes sense - but it would be great to point that out after the install.

balloob, 2021-06-18:
The ESP Web Tools website is a page aimed for developers to learn about the button. It's not actually meant to use as a place to install firmware. I think that such instructions would be great to add to places where it's used.

tcurdt, 2021-06-18:
I got this working on one nodemcu ... another one now is just stuck at "erasing device"

[image]

Any thoughts?

TD-er, 2021-06-18:
> I got this working on one nodemcu ... another one now is just stuck at "erasing device"
> 
> [image]
> 
> Any thoughts?

What kind of module is it?
NodeMCU with CP2102 chip?
Something like this one?
[image]

tcurdt, 2021-06-18:
I have have two. One works. One got/gets stuck during erasing.
[image]
[image]

TD-er, 2021-06-18:
Those have a CH340 USB to uart chip.
But the ESP module itself is one  not from Espressif, but the module is an AI-Thinker, which I believe is the same as I have. (only different revision)

So maybe it is a timing issue then.
Your board is also relatively old, as is the batch I am having troubles with.

tcurdt, 2021-06-18:
@TD-er If there is anything I can do to help to debug this - let me know.

TD-er, 2021-06-18:
I think @balloob has way more experience in this