esphome/esp-web-tools #27: ESP8266EX Support?

ghtomcat, 2021-06-23:
I can successfully flash my ESP32 but not my two NodeMCU (v2 and v3), which use the ESP8266EX Chip. 

error: "failed_initialize", details: Error: Unknown Chip

`esptool flash_id` returns:

```Serial port /dev/cu.usbserial-10
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c4:5b:be:60:dc:da
Uploading stub...
Running stub...
Stub running...
Manufacturer: d8
Device: 4016
Detected flash size: 4MB
```

TD-er, 2021-06-23:
I just tested it myself on a node that's not possible to flash using the web flasher:

[image]

```
esptool.py v3.0
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:3a:e8:39:13:7c
```

<html>
<body>

Flash Chip ID: | Vendor: 0x20 Device: 0x4016
-- | --
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT

</body>
</html>

TD-er, 2021-06-23:
I added a breakpoint to where the exception was caught:
`domexception a framing error has been detected`

conradopoole, 2021-06-23:
> I can successfully flash my ESP32 but not my two NodeMCU (v2 and v3), which use the ESP8266EX Chip.
> 
> error: "failed_initialize", details: Error: Unknown Chip
> 
> `esptool flash_id` returns:
> 
> ```
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: c4:5b:be:60:dc:da
> Uploading stub...
> Running stub...
> Stub running...
> Manufacturer: d8
> Device: 4016
> Detected flash size: 4MB
> ```

Can you post what you see on Chrome's Developer console?

conradopoole, 2021-06-23:
> I just tested it myself on a node that's not possible to flash using the web flasher:
> 
> [image]
> 
> ```
> esptool.py v3.0
> Serial port COM3
> Connecting....
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: 2c:3a:e8:39:13:7c
> ```
> 
> Flash Chip ID:	Vendor: 0x20 Device: 0x4016
> Flash Chip Real Size:	4096 kB
> Flash IDE Size:	4096 kB
> Flash IDE Speed:	40 MHz
> Flash IDE Mode:	DOUT

Is that all the console shows? No log messages?

TD-er, 2021-06-23:
Yep that's all.

conradopoole, 2021-06-23:
> Yep that's all.

Can you click on the login levels and enable Verbose?

[image]

TD-er, 2021-06-23:
Not much of a difference:
[image]

ghtomcat, 2021-06-23:
> Can you post what you see on Chrome's Developer console?

It's empty, even with Verbose enabled. Do i need to do anything to have more debug output like @TD-er above?

balloob, 2021-06-23:
By default it doesn't log to the console. I've just updated the demo to do so e6a198fefab3d16193bf797a51345716c96c82ab

ghtomcat, 2021-06-23:
> By default it doesn't log to the console. I've just updated the demo to do so [e6a198f](https://github.com/esphome/esp-web-tools/commit/e6a198fefab3d16193bf797a51345716c96c82ab). So now open the chrome dev tools prior to flashing.

Thanks, now i see the following in the console (with the demo)

[image]

TD-er, 2021-06-23:
[image]

This attempt I held GPIO-0 down and pressed reset.
The attempt before was the same, without reset + GPIO-0 held.

balloob, 2021-06-23:
The error has been updated by @conradopoole. It should now include the magic value of your chip so we can add it to the detected list. The demo has been updated. 

@ghtomcat can you try it again and report the updated error message.

ghtomcat, 2021-06-23:
> The error has been updated by @conradopoole. It should now include the magic value of your chip so we can add it to the detected list. The demo has been updated.

```start-flash-542f194f.js?module:1 Error: Unknown Chip: 0x-F3EFF```

TD-er, 2021-06-23:
Me too, on one module (with CH340) Unknown Chip: 0x-F3EFF

The unresponsive unit with CP2102 USB to serial:
[image]

balloob, 2021-06-23:
@ghtomcat this should be fixed in 3.4.2, live now in the docs. Can you try it?

TD-er, 2021-06-23:
Does work with the node that gave the unknown chip error before.

The "timeout" unit still remains unresponsive, but had not expected it to be fixed by these changes.

closed 2021-06-24 by balloob

ghtomcat, 2021-06-24:
> @ghtomcat this should be fixed in 3.4.2, live now in the docs. Can you try it?

Works now, thank you!

degraafm76, 2022-01-15:
> Me too, on one module (with CH340) Unknown Chip: 0x-F3EFF
> 
> The unresponsive unit with CP2102 USB to serial: [image]

Did you get the CP2102 working? if have the same issue with ESP8266EX modules

ghtomcat, 2022-01-17:
@degraafm76 yes, after the update it works for cp2102 and ch340

degraafm76, 2022-01-17:
> @degraafm76 yes, after the update it works for cp2102 and ch340

I see, but not for ESP8266EX modules.