esphome/esp-web-tools #52: Add multiple install buttons on the same page

m1cr0lab, 2021-08-15:
**First of all, thank you very much 💖 for this indispensable tool!**

How to put several buttons on the same HTML page?

By doing the following, the buttons appear normally on the page:

```html
<esp-web-install-button manifest="manifest_1.json"></esp-web-install-button>
<esp-web-install-button manifest="manifest_2.json"></esp-web-install-button>
<esp-web-install-button manifest="manifest_3.json"></esp-web-install-button>
```

During the first download (with any of the buttons), everything goes well:
the progress bar appears correctly during the whole installation process.

**But when you want to download again (with another button), the progress bar does not appear anymore.** 😕 

Is there anything special to do in this context, or is it a bug?

I thank you in advance for your help.
Steph

balloob, 2021-08-23:
That's a bug. We are using module-scoped variables which are the same for each instance.

Instead we should attach this info to the button.

The logic is here https://github.com/esphome/esp-web-tools/blob/main/src/start-flash.ts#L9-L15

closed 2021-08-24 by balloob

m1cr0lab, 2021-08-25:
Thank you very much for this fix!

PR #59 (2021-08-24): Store state on button
Store state in the button instead of globals to allow multiple buttons on a page.

Fixes #52