esphome/esp-web-tools #55: Old partitions.bin file

myhomeiot, 2021-08-21:
Please replace file `static/firmware_build/partitions.bin` with newer version, current version has smaller OTA partition sizes (0x190000 instead 0x1C0000) see https://github.com/esphome/issues/issues/1076
ESPHome already use newer version when flash using host USB.

gen_esp32part.py output for existing partitions.bin
```
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,1600K,
app1,app,ota_1,0x200000,1600K,
eeprom,data,153,0x390000,4K,
spiffs,data,spiffs,0x391000,60K,
```

gen_esp32part.py output for newer version partitions.bin
```
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,1792K,
app1,app,ota_1,0x1d0000,1792K,
eeprom,data,153,0x390000,4K,
spiffs,data,spiffs,0x391000,60K,
```

balloob, 2021-08-22:
People provide their own partition files. ESP web tools is just the tool to flash it. 

Do you mean to report this to the esphome dashboard?

myhomeiot, 2021-08-22:
It's uses old static partitions.bin which is a part of this repository :-( see [here](https://github.com/esphome/dashboard/blob/e31e95a077bdf034a5b384b37cd550a4aabba55b/src/flash.ts#L24)

I will open issue connected to this problem at dashboard repo, but if we will have new partiotion file here it's will almost solves the problem, because not many users use custom partiotions.

balloob, 2021-08-23:
It's not part of this repository, it's here https://github.com/esphome/dashboard/tree/main/raw_package/static/firmware

closed 2021-08-23 by balloob