esphome/esp-web-tools #85: Add Support for LOLIN S2 Mini

mawa85, 2021-10-31:
I tried to flash a LOLIN S2 Mini, but get the error:

Your ESP32-S2 board is not supported.

Board is this one
https://www.wemos.cc/en/latest/s2/s2_mini.html

If you need anything please let me know, happy to support testing.

youngd24, 2021-12-30:
+1 on this, have the same issue

jh-maker, 2022-01-28:
Any news on supporting this model? If anybody needs one to implement this model let me know.

witnessmenow, 2022-02-24:
If you update the manifest to be **ESP32-S2** rather than the ESP32S2 as described on the readme it seems to work

balloob, 2022-03-08:
Updated docs in 18c2053

In the future, if you notice an issue in the docs, please open a PR.

closed 2022-03-08 by balloob

ZanzyTHEbar, 2023-02-04:
i've tried this - it tells me that the board is not supported. 
My `manifest.json` file
```json
{
    "name": "OpenIris",
    "version": "2021.11.0",
    "new_install_prompt_erase": true,
    "builds": [
        {
            "chipFamily": "ESP32-S2",
            "parts": [
                {
                    "path": "merged-firmware.bin",
                    "offset": 0
                }
            ]
        }
    ]
}
```
Iam using an `ESPCam` board - funny thing is if i use `ESP32` - it works fine, nothing is wrong with the firmware. Setting it to `ESP32-S2` fails for some reason.

TD-er, 2023-02-04:
As far as I know, there is no ESP-cam board with an ESP32-S2, so it might be exactly according to specs what you're seeing here.

ZanzyTHEbar, 2023-02-04:
correct - they use the ESP32-S - i was conflating the versions. Thank you
for the sanity check :)

balloob, 2023-02-04:
Please no discussion on old issues.