esphome/esp-web-tools #95: Allow to scale layout of the install dialog

TD-er, 2021-11-10:
See the screenshot:
[image]

Also the "All existing data will be erased from your device" is a bit frightening. Let's hope it will only flash what I have set in the manifest.

balloob, 2021-11-10:
It will erase all data because it is unable to detect the firmware via Improv Serial, so it assumes it is a fresh installation.

TD-er, 2021-11-10:
Yep, I just found out... (on my dev/test node, so no big deal)
Can we disable this? or is that what you're selecting when configuring the improv setting in the manifest?

balloob, 2021-11-10:
The way it works now:

ESP Web Tools uses Improv Serial to detect the currently installed firmware + firmware version. If it can't detect it, it assumes it is a fresh installation.

balloob, 2021-11-10:
For your layout, any reason you're having the version be included in the name too? Shouldn't your name be "ESP Easy Mega" ?

TD-er, 2021-11-10:
Well we do have a lot of builds, so based on the date is standard for ESPEasy builds.
Would it be useful to change the underscores to spaces to not mess up the layout?

And can it be disabled to erase the flash and just flash it?
As of now, there is no Improv running on ESPEasy, so that means it cannot be used to update a node using this flasher, which cannot be updated via OTA. (e.g. just about any 1M node out there)

balloob, 2021-11-10:
I would put the other parts in the name. Keep the version out.

balloob, 2021-11-10:
And yes use spaces.

TD-er, 2021-11-10:
[image]
With a little `sed -i` calls on the command line, I updated all the manifest JSON files, to remove the .bin from the name and changed "ESP_Easy_mega_20211105_" into "ESP Easy mega "
So that does indeed make it a bit more readable.
Will later think of a better naming schema for it.

balloob, 2021-11-10:
Since ESP8266 and ESP32 can be in the same manifest (they are automatically detected), you can drop that part of the name too.

TD-er, 2021-11-10:
Well not all builds have an equivalent for ESP8266/ESP32.

And about the erase part.. I have been looking for the Improv serial documentation, but really can't find it :(

balloob, 2021-11-10:
Sorry, yes, the website is a little behind, we will publish it later today <https://deploy-preview-7--improv-wifi.netlify.app/serial/>

TD-er, 2021-11-10:
OK, so now I can at least understand what this message means:
```
Writing to stream: [0x49, 0x4D, 0x50, 0x52, 0x4F, 0x56, 0x01, 0x03, 0x03, 0x02, 0x00, 0x02]
```
But it is not yet clear what to reply to it to make sure the device will not get erased. And like I said, when updating existing nodes you cannot rely on the presence of such IMPROV support.

balloob, 2021-11-10:
Here is the ESPHome implementation https://github.com/esphome/esphome/pull/2423

balloob, 2021-11-10:
I've merged the docs now with latest version (1 commit was missing). Should all be live now.

balloob, 2021-11-11:
Closing this issue as the layout issue was resolved by replacing underscores with spaces.

closed 2021-11-11 by balloob