esphome/esp-web-tools #96: Commands sent to the console not sent to the ESP

TD-er, 2021-11-10:
Just tried the new  "Logs" console.
[image]

Really useful and looking nice!
But.. as you can see in the screenshot, I tried to send a few commands to it, but those do not seem to be processed.
Are the commands, or whatever you type to it, being forwarded to the ESP?
And if so, what is used as the line termination?

What does the "send" button do? 

Also after a while (only a few minutes) I got this Framing error:
[image]
So I guess the terminal session somehow got interrupted and does not continue.

balloob, 2021-11-10:
Ah, right now we don't use any line ending when sending a command. That's not good.

https://github.com/esphome/esp-web-tools/blob/main/src/components/ewt-console.ts#L184

Would you expect `\r\n` ?

TD-er, 2021-11-10:
A CRLF is fine, I only check for one and ignore the other, so CRLF is OK.

balloob, 2021-11-10:
What is convention? Would be good to do the right thing.

balloob, 2021-11-10:
Fix here https://github.com/esphome/esp-web-tools/pull/97

closed 2021-11-10 by balloob

TD-er, 2021-11-10:
There is still some `IMPROV` in the command buffer for the first command given:
[image]

TD-er, 2021-11-10:
Also it is unclear what the "Send" button does or should do

balloob, 2021-11-10:
The Send button does the same as pressing ENTER

TD-er, 2021-11-10:
> The Send button does the same as pressing ENTER

Nope, not here in my setup.

PR #97 (2021-11-10): Finish sending commands with \r\n
Fixes #96