esphome/esp-web-tools #109: Serial port is inaccessible right after flashing

TD-er, 2021-11-11:
[image]

Right after flashing another dialog is shown very briefly, followed by this one.

N.B. something similar happens if you had the "logs/console" dialog open and that one looses connection. Sometimes you then have to wait quite a while (or reload the page) to be able to access the device again.

balloob, 2021-11-11:
In 4.1.0 I’ve added an extra 100ms delay after writing is done + reset before trying to re-detect Improv Wi-Fi. Is this 4.0.0 or 4.1.0?

TD-er, 2021-11-11:
It is the latest at the time I added the issue, so 4.1.1 I guess.

balloob, 2021-11-16:
Your error message shows "A break condition has been detected". What is a break condition?

TD-er, 2021-11-16:
It did reboot.
So maybe some sent character did not complete?
The USB connector was not removed.

N.B. maybe this is also related to the lost connection I sometimes see in the log dialog.