esphome/esp-web-tools #111: [Feature] Allow to differentiate based on flash size

TD-er, 2021-11-11:
Builds can now be defined based on `chipFamily`, but it would also be nice to differentiate based on flash size.

For example:

```json
{
  "name": "ESPHome",
  "version": "2021.10.3",
  "home_assistant_domain": "esphome",
  "builds": [
    {
      "chipFamily": "ESP8266",
      "flashSize": "4M",
      "parts": [
        { "path": "esp8266_4M.bin", "offset": 0 },
      ]
    },
    {
      "chipFamily": "ESP8266",
      "flashSize": "1M",
      "parts": [
        { "path": "esp8266_1M.bin", "offset": 0 },
      ]
    }
  ]
}
```

TD-er, 2021-11-11:
I would say "exactly" match.
But on the other hand, if we don't have like an 8M build, but do have an 4M and 16M, then the 4M would be the best match.
However if a user would accidentally flash the wrong version, the file system would be deleted so maybe best to offer a dialog then suggesting there is no matching build, maybe offering to continue at the users own risk.

Aircoookie, 2021-11-16:
I'd second the JSON structure proposed by @TD-er.

For choosing which build to install, it ideally should be the build which exactly matches the flash size, but if no such build is available, it should install the one intented for the closest smaller size.
On clean install (either no Improv detected or a different firmware) we don't have to worry about versions that much since there is no data to be preserved anyways.

On update, I see how a warning dialog could be useful, but in that case we don't know at all what partition layout the user is using, that can't reliably be detected using the auto-detected size alone. Perhaps we should add an optional flash size string to Improv... but then again, you could still e.g. use different partition tables both with a size of 4MB total - with 1.5mb FS or 1mb FS size. Of course, we could pass a string like `4M,1M`, denoting overall target flash size and FS size, but I don't know if it'll be worth it.

And there should definitely be an option to specify a default build per chipFamily in case auto detecting the size is not possible (although I have yet to observe such a case)

TD-er, 2021-11-16:
@Aircoookie In my ESPEasy builds, I have in all builds (except for the ESP8285 or special hardware builds) the flash layout specified in the file name.

- 4M1M
- 4M2M
- 4M318k  (for ESP32, but that's the only layout I use for 4M on ESP32)
- 16M1M (SPIFFS)
- 16M2M (LittleFS)
- 16M8M (LittleFS)

etc.

Thus specifying the file system en partitioning would be also nice to have.
But that's only diable when adding improv.

balloob, 2021-11-16:
Most users will not know what size their FS has. When users see the word "Update" they definitely feel like it's save to update. It shouldn't end up being incompatible.

Flash size can be detected by rebooting the ESP into the bootloader. Some devices require the BOOT button to be held, so we only do that when a user clicks install/update. This means we cannot use this information in the dashboard unless Improv tells us about this information.

TD-er, 2021-11-16:
Differentiating on flash size is already a big improvement if it is made possible.
For example it will help users to get started with at least working defaults.
On a clean unit there is for sure no improv running, so that shows the importance of at least some flash size detection.

If we can already use the most commonly used defaults for this, then users may only be left with the choice of file system. But I think that's also going to become an obsolete choice very soon. Here improv can help, so I will try to add it ASAP.

Right now Improv Serial has documented this:

> Order of strings: Firmware name, firmware version, hardware chip/variant, device name.
> 
> Example: ESPHome, 2021.11.0, ESP32-C3, Temperature Monitor.

Is it possible to define 1 or 2 more fields?
- preferred flash layout
- File System

I do understand that file system detection is not the responsibility of the flasher, thus when no improv is used then it will just be the best default. But if improv is present, it would be nice to get a build matching whatever is already present.

Aircoookie, 2023-01-11:
Might be time to revisit this. As a first step towards implementing this, I propose we add flash size to the device informaton RPC result in Improv Serial.
As for the format, probably going with the ESP custom of a string like `4M1M` for 4MB flash with 1MB FS would be easiest.

balloob, 2023-01-16:
The problem with adding it to Improv is that Improv is only available on upgrade.

lost-hope, 2024-03-11:
I guess that there haven't been any changes to this right? 
I see that there is something about Detected Flash size in the installdialog.js.