esphome/esp-web-tools #112: [Feature] Allow to detect ESP8285 (if possible)

TD-er, 2021-11-11:
Not entirely sure if it is possible, but it would be great if the ESP8285 could be detected as chip family.

At runtime I have not found a way to distinguish the ESP8285 from an ESP8266, so I am not sure if it is possible to detect when probing the ESP.

balloob, 2021-11-11:
We detect ESP Type by reading a value from the memory

https://github.com/NabuCasa/esp-web-flasher/blob/main/src/const.ts#L122

And then map those values to ESP chip families

https://github.com/NabuCasa/esp-web-flasher/blob/main/src/const.ts#L140

If the ESP8285 is different, we could detect it

TD-er, 2021-11-11:
The main difference, as far as I know, is that the ESP8285 does allow to use 2 more GPIO pins, which an ESP8266 can't as it needs it for the flash.
But so far I have not seen an option to detect it in the chip itself.
I will later this evening have a look at the reported value there.

balloob, 2021-11-15:
This is what `esptool` can detect https://github.com/espressif/esptool/blob/master/esptool.py#L381-L383

balloob, 2021-11-15:
And this is how they detect esp8285 https://github.com/espressif/esptool/blob/master/esptool.py#L1313

We should be able to add that to https://github.com/NabuCasa/esp-web-flasher/

TD-er, 2021-11-15:
Great!
Would be nice if I can also detect this in runtime. Will check it too.

N.B. I have never seen the `ESP8285H16` being used anywhere. I know there is an ESP8285 with 2M flash, but I only found it in datasheets.

TD-er, 2021-11-15:
Ah found an issue describing this: https://github.com/esp8266/Arduino/issues/7340

balloob, 2021-11-15:
Why would you need to detect it in runtime? You can just feed it a different build and have the chip family be embedded in the build.

balloob, 2021-11-15:
btw looks like ESP Web Flasher had support for ESP8285 but commented out … https://github.com/NabuCasa/esp-web-flasher/blob/main/src/esp_loader.ts#L121-L125

TD-er, 2021-11-15:
> Why would you need to detect it in runtime? You can just feed it a different build and have the chip family be embedded in the build.

As far as I can tell, the builds for ESP8266 and ESP8285 only differ in the selection of which GPIO pins can be used.
So if I can detect it at runtime, I can get rid of ESP8285 builds, which does save in the number of builds.
But just in case I'm wrong in my assumptions, we still need the flasher selection ;)

balloob, 2021-11-15:
Can't you do the same as ESP Tool does and read the fuses?

TD-er, 2021-11-15:
That's what I intend to do, now that I know where to look.