esphome/esp-web-tools #113: No error dialog when com port cannot be opened

TD-er, 2021-11-11:
[image]

The com port is locked (I opened a Putty session to reproduce it) and pressed "Install"
Then accepted the pop-up to connect to the COM port and nothing seems to happen.
The error is present in the console, but no dialog is shown to indicate to the user what went wrong.

balloob, 2021-11-11:
Ah snap. We filter out errors at that point in time because the exception usually is because users pressed “Cancel” instead of picking a port. I’ll take a look if we can differentiate the errors.

balloob, 2021-11-11:
Looks like `AbortError` should be treated differently https://developer.mozilla.org/en-US/docs/Web/API/Serial/requestPort#exceptions

PR #115 (2021-11-12): Do not silently ignore errors not caused by user closing picker
`await navigator.serial.requestPort()` will raise not just when the user clicks "cancel", on Widnows it can also raise when the serial port is already in use.

Fixes #113
closed 2021-11-12 by balloob