esphome/esp-web-tools #118: Error! Failed to initialize.

vicfixxx, 2021-11-14:
When i click the install button in the WLED application, i get the following response:

Failed to initialize. Try resetting your device or holding the BOOT button while selecting your serial port.

I have a ESP8266 chip and i also tried ESP home and Tasmota, both gave the same issue.

balloob, 2021-11-14:
What device?

Does it have a BOOT button?

Did you hold it while pressing install?

vicfixxx, 2021-11-14:
Hi,

I use the AZDelivery D1 Mini NodeMcu met ESP8266-12F and unfortunatly it doesn't have a BOOT button.

If i try to install something different, it works. I also tried with a brand new device, and get the same error message.

TD-er, 2021-11-14:
Does the serial port show up in the task manager? And is it still present when the failure appeared?
I've seen a few of those that had really flaky USB connectors.
You may also try a different USB cable and/or connect it directly to an USB port on the PC instead of via a hub.

vicfixxx, 2021-11-14:
I can see the device in my device manager with a COM port (currently COM 4).

I've tried a couple of USB cables and i can see the device in the device manager, so that should be good.

I've attached the specific error on the webpage.

The funny thing is that i've tried on a different computer and i get the same error.

Should i check the COM settings, like BAUT rate? if so, do you know what data i have to use?

Thanks in advance ;)

[image]
.

vicfixxx, 2021-11-15:
Hi, I probably applied too much heat to the board during soldering, so I've ordered some new ones and try it out on those. As soon as i know more, i'll let you know 😎

balloob, 2022-03-08:
Closing this issue as this is based on an old version of ESP Web Tools.

closed 2022-03-08 by balloob

adman120, 2022-03-17:
Can you reopen the issue, it is still happening, on current version for me, with same model esp.

balloob, 2022-03-17:
Please open a new issue and include logs + model and steps you did to reproduce.

adman120, 2022-03-17:
Will do, no problem.

Oinq, 2022-11-23:
Where did u opened it? I'm facing the same described above with the same hardware.

finnfreijer, 2023-02-12:
hallo ik heb een esp32 en ik probeer wled er op te downloaden maar elke keer als ik dat probeer geeft die aan dat ik de boot knop moet in drukken maar als ik dat doe verandert er niks. wat moet ik doen???

MichalKrasowski, 2023-07-24:
In my case it was lack of privileges for `/dev/ttyUSB0`.

Granting access to all ( `sudo chmod 777 /dev/ttyUSB0`) proves it's privileges issue.

Relevant devtools log on failure:
`:673  DOMException: Failed to execute 'open' on 'SerialPort': Failed to open serial port.
_handleConnectSerialSubmit @ wi`

phanky5, 2023-08-17:
Thank you @MichalKrasowski this solved the issue for me.

Id2ndR, 2024-04-22:
Granting access to other (`sudo chmod o+rw /dev/ttyUSB0`) temporary solve the privileges issue. The 'x' permission is not required.

To permanently solve this, you must add yourself in the group associated to this `/dev/ttyUSB0` file.
This can be done with the following command followed by a restart of you session: `sudo usermod -a -G $(stat -c %G /dev/ttyUSB0) $(whoami)`

skyfly200, 2026-01-15:
Im having this issue now with my ESP32-S2 on a windows machine. Not sure what to do since I dont have the linux env to try this solution.