esphome/esp-web-tools #130: Bring back the user feature of erase flash

arendst, 2021-11-24:
For Tasmota we prefer NOT to erase flash on subsequent use of esp-web-tools. This way the code can be updated while keeping the settings.

Used to work in the 3.x release but now only works when configured at manifest level.

balloob, 2021-11-24:
Can you define "user"?

There are currently two ways when installing firmware when it won't erase the flash:
- Any installation if `new_install_skip_erase` is set to `true` in the manifest
- Implement Improv Serial and the firmware in the "Device Information" response matches the manifest name

arendst, 2021-11-25:
The "user" in this case the physical user selecting the option below

[image]

Where the html is:
```
      <p>
        <input type="checkbox" id="erase" name="erase" onchange="toggleErase()">
        <label for="erase">Reset settings to default</label>
      </p>
```
and the underlying javascript looks like this:
```
      function toggleErase() {
	    if (document.getElementById('erase').checked) {
	      document.getElementById('inst').setAttribute('erase-first','');
	    } else {
	      document.getElementById('inst').removeAttribute('erase-first');
	    }
      }
```

balloob, 2021-11-25:
The way ESP Web Tools works has changed.

Previously the button _was_ the UI and via attributes a user could create their own UI around it.

In the new version there is a dialog prompting the user what to do. It no longer immediately starts installing the firmware. It is a bad user experience if a user can check a checkbox to "Reset settings to default" to get to the console to enter some commands.

Without Improv Serial:

[image]

With Improv Serial:

[image]

arendst, 2021-11-26:
I understand.

I think I'll stick with 3.x as I need the erase option BEFORE flashing. Using a terminal to reset settings that made the flashed device unusable (like broken serial code in the ESP32 SDK/IDF/Core does) won't work.

Thx.

closed 2021-11-26 by arendst

balloob, 2021-11-26:
I would love to better understand your use cases.

I just noticed that your [upgrade instructions](https://tasmota.github.io/docs/Upgrading/#serial-upgrade) contain a serial section that points at your installation website.

Is it correct that you're trying to support the following use cases:

1. a new device with unknown existing content doing a fresh install
2. a device running an older version of Tasmota doing an upgrade over webserial

I personally think that use case 2 is a very small group because the OTA update option is far easier. The added benefit of the console (which I added partially because one of your team members [+1d the request](https://github.com/esphome/esp-web-tools/pull/87#issuecomment-961748958)) allows users to configure Tasmota without having to go through the painful Soft AP steps to connect it to Wi-Fi. If you would add Improv Serial support the experience could be even better.

TD-er, 2021-11-26:
> I personally think that use case 2 is a very small group because the OTA update option is far easier.

Not sure how big the binaries of Tasmota are these days, but I know that for ESPEasy users it is becoming quite a struggle to OTA on 1M units. (next to impossible)

balloob, 2021-11-26:
@TD-er @arendst if we change the `new_install_skip_erase` option to `new_install_prompt_erase`, and if set to True we prompt users if they want to erase their device before installing the new version, would that help?

TD-er, 2021-11-26:
For ESPEasy it would help for sure as there can be valid reasons to perform a full wipe, even when the ESP seems to run fine and even if Improv (when implemented, working on it) does report there is a version running that needs only an upgrade.

The problem is that sometimes, for whatever reason, the WiFi doesn't work stable or at least not as stable as it could.
The only remedy then is to completely wipe the flash and start all over.

If there is a toggle possible to force a wipe, but having it disabled by default, then that would be great.
Or maybe a button (in a separate color) in the console screen would also be useful I guess as the console is more like the "expert" mode.

arendst, 2021-11-26:
Yes it would help but we still have to update all manifests. Perhaps the default can be prompt for erase if the parameter is not present

Jason2866, 2021-11-29:
@balloob the +1 for Terminal window was for seeing what happens when ESP starts with flashed firmware. For Tasmota users most of the time the assigned IP address is from interest. Power users do additional individual configs when flashing from scratch.
Many Tasmota users like to UPDATE via esp-web-tools there devices without erase flash. We want to provide as many ways as possible to update and flash devices.
Swapping different firmware builds with esp-web-tools and keeping settings (as it is in the old version) is so easy and works like a charm

balloob, 2021-11-29:
The terminal window has a "RESET DEVICE" button which offers exactly that feature 👍

TD-er, 2021-11-29:
Isn't "Wipe Flash" or "Format" a better description?
"reset" can also be interpreted like pressing the reset button.
Been there, where I told someone to restart a router, but they found a reset button on it and even though it took them a while to find a pin small enough and also a few retries, they managed to wipe the entire configuration where only a reboot was needed.

So I do have developed some kind of allergy for the word "reset"...
If the button is meant to do something different than wiping, then I guess I also made my point ;)

balloob, 2021-11-30:
PR implementing `new_install_prompt_erase`: https://github.com/esphome/esp-web-tools/pull/134

sblantipodi, 2022-04-17:
> The terminal window has a "RESET DEVICE" button which offers exactly that feature 👍

@balloob you closed this issue but you haven't solved it really :)
the "RESET DEVICE" button you are referring, only REBOOTS the microcontroller, does not reset it ("Wipe Flash" or "Format") like @TD-er and others was trying to explain you.

I'll put this features on vote. I think that a "Wipe flash" option should be always visible to user before installing the firmware.
No matter if it's same name or version.  😄 

With previous version this is possible, why insert a regression in the newer version of esp web tools?