esphome/esp-web-tools #135: Should be able to set timeout for improv response after flash

amandel, 2021-12-04:
Thanks for developing improv and the eco system around this.

Our [OpenBikeSensorFirmware](https://github.com/openbikesensor/OpenBikeSensorFirmware) takes quiet a bit of time during the initial startup (ssl cert is created, wifi connection is looked up, etc.). So the 1 or 10 seconds the esp-web-tools are waiting for  the 1st improv message are not enough. 

Could the timeout be exposed as an option? 

https://github.com/esphome/esp-web-tools/blob/0266057a383aa30e5ab0b33efb930af1eb41a55c/src/install-dialog.ts#L703

balloob, 2021-12-06:
Is even 10 seconds not enough? 😱 

My initial idea was to just always wait 10s https://github.com/esphome/esp-web-tools/compare/always-delay

amandel, 2021-12-06:
Unfortunately not I did not measure the time (will do) but creating the SSL certificat can take a minute.

closed 2021-12-07 by balloob

amandel, 2021-12-08:
Thanks a lot for adding this.

PR #139 (2021-12-07): Allow configuring improv timeout
Add support for a new `new_install_improv_wait_time` that allow you to specify the time ESP Web Tools should wait for an Improv Serial response after installing the firmware (default is 10 seconds). Set to 0 to disable waiting for Improv. 

@amandel can you verify that this build would solve your problems? 

Fixes #135