esphome/esp-web-tools #136: [Feature] Expose serial data while flashing / bootup in a detail view

amandel, 2021-12-04:
For debugging or users interested in details it might be useful to have the option to expand a "detail" pane during the install steps. The detail pane could show data received via serial from the connected device at that time.

This could help in error situations where a device can not be reset or shows errors during boot after the flash process. 

I remember a similar thing when updating Ubuntu packages via UI here you can look at the console data in such a - initially hidden - detail pane.

balloob, 2021-12-06:
Ubuntu UI will actually run the CLI tools in the background. This is not the case here, it is fully implemented in JavaScript. There is no log output.

amandel, 2021-12-06:
Oh, my hint to Ubuntu was just to point out how it could look like.
Content should be the serial data received from the ESP.

balloob, 2021-12-06:
We're not using the firmware but are using a bootloader to install the data.

amandel, 2021-12-06:
I guess we are still not talking about the same thing here? I'm sorry for being so unspecific. Might be also that the data is technically not available via the interface you use. Listening on serial during flash I see the following data:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
À  UU    ÀÀ  UU    ÀÀ  UU    ÀÀ  UU    ÀÀ  UU    ÀÀ  UU    ÀÀ  UU    ÀÀ  UU    ÀÀ
...
  ÀÀ       ÀÀ       ÀÀ       ÀÀ       ÀÀ       ÀÀ       ÀÀ       ÀÀ       Àets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][OpenBikeSensorFirmware.cpp:222] setup(): openbikesensor.org - OBS/v0.10-dev
...
```

This might be useful for debugging or interesting for users that want to see every detail. Might be it is not accessible to you as easy as I initially thought and you can only read the data in the "Logs & Console" mode.

balloob, 2021-12-17:
We do show some information in the JavaScript console

[image]