esphome/esp-web-tools #148: How to properly set the manifest "parts" from a PlatformIO project

zakaton, 2021-12-20:
We're using PlatformIO for [our ESP32-based firmware](https://github.com/Ukaton-Inc/side-missions-firmware/tree/ble), and would like to know how to properly fill out the "parts" part of the manifest for our project, as well as which files we'd need to include.

When we build we can see the files at .pio/build/nodemcu-32s/, but which files do we include and how do we get the right offsets? We also have a [partitions.csv](https://github.com/Ukaton-Inc/side-missions-firmware/blob/ble/partitions.csv) file, but we don't see the equivalent files in the build section.

Sorry for being so ignorant about this kinda stuff, but I'd really appreciate the help because being able to update our firmware on the web would be so cool and useful!

jesserockz, 2021-12-20:
Hi @zakaton
The required binary files along with their offsets for an environment are listed in the platformio idedata json structure (`platformio run -t idedata`).

Jesse

zakaton, 2021-12-20:
I ran the command, and under extras/flash_images I found 3 files with their offsets (bootloader_dio_40m.bin, partitions.bin, boot_app0.bin), and added them to the manifest, but it still didn't work.

Then I looked at the manifest example in the readme and looked for an esp32.bin, which I finally realized referred to the firmware.bin in the build folder, and assumed the offset was 65536 just like in the example, and [now it works](https://twitter.com/ConcreteSciFi/status/1473071309315661825?s=20)! Thank you so much, this is so cool!

closed 2021-12-20 by zakaton