esphome/esp-web-tools #151: Bug when using ESP32 with Arduino ESP32 V2.0.2 (based on esp-idf v4.4)

wladwnt, 2021-12-30:
Flashing using esp tool is OK (esptool.exe --chip esp32 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0xe000 boot_app0.bin 0x1000 Programm.ino.bootloader.bin 0x10000 Programm.ino.bin 0x8000 Programm.ino.partitions.bin)

But when esp web tools are used (V6.1.1, V5.3.0 tested too) the endless reset loop is shown in terminal log:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
```

Manifest file used:
```
{
  "name": "Program",
  "version": "11",
  "builds": [
    {
      "chipFamily": "ESP32",
      "parts": [
        { "path": "Programm.ino.bootloader.bin", "offset": 4096 },
        { "path": "Programm.ino.partitions.bin", "offset": 32768 },
        { "path": "boot_app0.bin", "offset": 57344 },
        { "path": "Programm.ino.bin", "offset": 65536 }
      ]
    }
  ]
}
```

TD-er, 2021-12-30:
I am very curious to this one, as I do get this with my own development building ESP32 v2.0.2.
So I guess it may have to do with wrong bootloader file, incorrect offset, etc.

wladwnt, 2021-12-30:
Well, I use exactely the same files and offsets as esptool uses. It must be difference inthe flash process, I suppose...

Jason2866, 2021-12-31:
Have you tried flash mode dout (with the corresponding bootloader file)?

wladwnt, 2021-12-31:
I have identifiend the problem a little bit: In the bootloader.bin file, that is flashed to addr 0x1000, the fourth byte is 0x2F. When I am flashing using esptool or ESP32 downloader, it is also 0x2F in the flash (at addr. 0x1003, check reading back with esptool). When flashing using web tools, it is 0x2F.
Double check:
1) flashing using web tool -> not OK
2) then flashing just the bootloader at addr. 0x1000 using ESP32 downloader --> now it works.

So it seems that web tool somehow change this byte from 0x2F to 0x20 while flashing.

TD-er, 2021-12-31:
Can you also check the file you're flashing?
What if ESP32 downloader does set a bit there based on the flash mode, where the web flasher does just copy it?

wladwnt, 2021-12-31:
Of course I have cheked it, the oroginal file has correct 0x2F.

TD-er, 2021-12-31:
The really strange thing here is that I'm experiencing similar behavior when flashing from my IDE and when I flash the built file via Espressif's downloader tool. But only if I work with 16M images. (or built for 16M flash)
If I flash a build made for 4M flash it works just fine, regardless whether I flash it to a 4M or 16M flash unit.

This is happening since Arduino ESP32 v2.0.2
So it seems as if there is something funky with the boot sequence in this newer Arduino ESP32 config and I have no idea anymore where to look for.
I can still try to flash it using the IDE => will boot-loop 
And then flash the bootloader file as you did.

Jason2866, 2021-12-31:
Just installed Tasmota32-OdroidGo via WebInstaller https://arendst.github.io/Tasmota-firmware/ Deleted flash, device started with no problem. It is a 16Mbyte flash device and 10Mbyte is used with LittleFS

wladwnt, 2021-12-31:
OK, it works, for example, when compiled with DOUT and 40 MHz and the respective bootloader is used (it has already 0x20 in the fourth byte).

Jason2866, 2021-12-31:
Yes, using the bootloader with the same flash mode and speed where the firmware is build is necessary.

balloob, 2021-12-31:
ESP Web Tools uses ESP Web Flasher to flash the image. There is some image patching logic in there: https://github.com/NabuCasa/esp-web-flasher/blob/main/src/esp_loader.ts#L769-L844

wladwnt, 2021-12-31:
Thank you, now everything make sense: ESP Web Flasher alsways change the fourth byte to 0x20 (dio, 40 MHz). Apparently Arduino-ESP32 V2.0.2 based firmware do not accept this change and can not be executed. If it is already based on bootloader tat has 0x20, then nothing is changed by ESP Web Flasher during flashing process and it works fine.

The question is: why the ESP Web Flasher do this change? The code says "For now we always select 40m, a common value supported by many flash chips and ESP boards" and it was probably ok for arduino-ESP32 V1.0x based code, but is not OK for Arduino-ESP32 V2.0.x based code.

Jason2866, 2022-01-02:
Patching flash mode and speed is ALWAYS https://github.com/NabuCasa/esp-web-flasher/blob/main/src/esp_loader.ts#L817-L820 a bad idea (imho a bug).
It cripples devices AND there are devices which are NOT working with mode DIO.
The mode which works for all is only DOUT.

balloob, 2022-01-02:
We didn’t write the flashing code. We ported the Adafruit code to be able to re-use it in other NPM packages (this one!) and added compression. I don’t understand that part of the code and would need in understanding what needs to be changed.

ESPFlasher doesn’t have this option. How do other flash tools do this?

wladwnt, 2022-01-02:
I think Adafruit has introduced this bin file modification because they know their hardware. The (probably most original) tool for ESP32 from espressif (ESP32 flash downaload tool) has basically two options (selectable by a checkbox):

- either to change bin files according to the chip capabilities detected (but I dont know what exactely they change),
- or not to change anything in bin files (=the user is responsbile to use the hardware that is compatible to the bin files)

I think this second option would be the best for a tool that is general and is not intended for a dedictaed hardware. --> Just delete these lines from teh code.
The firs option can be basically introduced (ba a kind of option), the condition is however that we would know exactely what and how to change in bin files. The present code does not do it apparently in a right way, generally.

TD-er, 2022-01-02:
Espressif's "download tool" does have a checkbox to not change the things you try to flash.

Jason2866, 2022-01-02:
yes, fast fix, remove the patching.

TD-er, 2022-01-02:
[...]

> I think this second option would be the best for a tool that is general and is not intended for a dedictaed hardware. --> Just delete these lines from teh code. The firs option can be basically [...]

Not 100% sure that is the best option, as it seems there is also some dependency on the used crystal.
At least on ESP8266 the crystal can be 26 or 40 MHz. It looks like this may also be true for ESP32, but so far I have not seen (or not remember to have seen) 26 MHz crystals.

wladwnt, 2022-01-03:
There is the information from espressif ([https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html))

Now it is clear why the behaviour is different for Arduino-ESP32 V2.0.2 (esp-idf 4.4) based SW and Arduino-ESP32 V1.0.x (esp-idf 3.x) based SW. This is also a clear argument to remove this code (because just changing the boatloader header is meaningles since esp-idf V4.0).

"Each ESP-IDF application or bootloader .bin file contains a header with CONFIG_ESPTOOLPY_FLASHMODE, CONFIG_ESPTOOLPY_FLASHFREQ, CONFIG_ESPTOOLPY_FLASHSIZE embedded in it. These are used to configure the SPI flash during boot.

The First stage bootloader in ROM reads the Second stage bootloader header information from flash and uses this infomation to load the rest of the Second stage bootloader from flash. However, at this time the system clock speed is lower than configured and not all flash modes are supported. **When the Second stage bootloader then runs, it will reconfigure the flash using values read from the currently selected app binary’s header** (and NOT from the Second stage bootloader header). This allows an OTA update to change the SPI flash settings in use.

**Bootloaders prior to ESP-IDF V4.0 used the bootloader’s own header** to configure the SPI flash, meaning these values could not be changed in an update. To maintain compatibility with older bootloaders, the app re-initializes the flash settings during app startup using the configuration found in the app header."

TD-er, 2022-01-03:
That does explain A LOT!!
Thanks!

closed 2022-01-03 by balloob