esphome/esp-web-tools #152: Automatically download correct bootloader based on selected firmware

balloob, 2021-12-31:
@Jason2866 wrote:

> A nice feature would be: WebTools scans firmware (magic first bytes https://github.com/esphome/esphome-flasher/blob/main/esphomeflasher/common.py#L143-L153) which Bootloader file and other files are needed and downloads it from espressif github repo. There is no need to provide this file from user side. Only partitions file is specific
Example for https://github.com/esphome/esphome-flasher/tree/main/esphomeflasher

balloob, 2021-12-31:
Websites cannot randomly download files from other websites and process the content due to CORS security restrictions. Our ESPHome Flasher doesn't have that restriction because it's in Python.

Since each project defines their own manifest to use ESP Web Tools, they are also able to include the other files.

Maybe someone could make a CLI tool to generate a manifest based on an ESP32 firmware and download the right files?

TD-er, 2021-12-31:
Or the files should just be included in a single file, just as I do for ESPEasy.
https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/post_esp32.py

It is just included in the PlatformIO build script.

Jason2866, 2022-01-02:
@TD-er I "borrowed" your platformio script. Will try to use with web-tools.
Every flash and speed option should be possible this way.

Jason2866, 2022-01-02:
@balloob  wrote

>Websites cannot randomly download files from other websites and process the >content due to CORS security restrictions in the browser. Our ESPHome Flasher doesn't >have that restriction because it's in Python.

Since the files are always the same. It must not be a random website where to download. I am a total Web coding and restrictions noob. Is there no way to download files from a hardcoded repo?  
User specific file is only the partition.bin. This file needs to be provided in any case from the user as the firmware. All other files are static

balloob, 2022-01-02:
We can only download files from the same host. It means you have to host them easier. I think it’s better if ESP Web Tools not dive into knowing how to find the right files and bundle them. That is up to the user to gather the right files to install.

Open a new issue for feature requests.

jesserockz, 2022-01-05:
> Or the files should just be included in a single file, just as I do for ESPEasy. [letscontrolit/ESPEasy@`mega`/tools/pio/post_esp32.py](https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/post_esp32.py?rgh-link-date=2021-12-31T22%3A26%3A13Z)
> 
> It is just included in the PlatformIO build script.

I have just added this to ESPHome via https://github.com/esphome/esphome/pull/3003

TD-er, 2022-01-05:
I just linked in that PR to the latest version of the script, now actually working on flash chips other than 4M in size ;)  (or at least with partitions extending the 4M limit)

balloob, 2022-01-16:
Going to close this. As it’s not ESP Web Tools responsibility. It could be a responsibility of an ESP Web Tools manifest creator tool, but that’s out of scope for this repo.

closed 2022-01-16 by balloob

Jason2866, 2022-01-17:
100% fine with closing, since it is possible and working to flash a one file firmware
(since no file patching is done anymore).

sblantipodi, 2022-03-16:
@TD-er sorry if I resume this old issue but I found it very useful, your solution with platformio is smart. :)
I can create the factory firmware and I can flash a single binary file with ESP Web Tools no problem but I have problems when I try to flash that firmware via OTA.

ESP32 complains about a "Wrong Magic Byte"...

Any idea on how to fix this using your POST script?

Jason2866, 2022-03-16:
@sblantipodi You can not flash via OTA the factory firmware. You have to use the firmware without `factory` in the name

Jason2866, 2022-03-16:
No, since the OTA part is Arduino core. If you want a workaround you have to change the OTA code there.