esphome/esp-web-tools #156: Problems flashing (chinese "Amica") NodeMCU with CP2101

degraafm76, 2022-01-05:
I have a couple of (chinese "Amica") NodeMCU with CP2101 that i cannot flash with ESP web tools. 
[image]

Flashing the same NodeMCU's with Arduino IDE or PlatformIO is working fine:

[image]

I allready tried different browsers/computers

balloob, 2022-01-05:
Could it be the baud rate? 

ESP Web Tools defaults to 115200 baud rate: https://github.com/esphome/esp-web-tools/blob/main/src/connect.ts#L21

degraafm76, 2022-01-06:
While the bottom of the NodeMCU states 9600 baud rate the upload speed in arduino IDE and PlatformIO is 115200 and working fine. I changed the baud rate to 9600 in esp web tools but uploading doesnt work. Openening the serial console through esp webtools does work , so the serial connection seems to be ok.

I browsed through the issue list and found https://github.com/esphome/esp-web-tools/issues/27. It seems there is a problem with uploading to ESP8266EX. This chip is on the boards i have. In this issue there are 2 problems described and it seems uploading to ESP8266EX is never solved.

If helpfull I can send you a board.

ldijkman, 2022-05-09:
Hello 
i have 2 of these nodemcu boards = ESP 12E
no problem with my esp web tools installer https://ldijkman.github.io/randomnerd_esp32_wifi_manager/Touch_Electra-Electra_Touch/

also have wemos uno sized ESP 12F wich also works ok installing

today i recieved 5 wemos d1 mini ESP12F and that does not work
looks like the code bin part is done at 26% and then the littlefs part "Timed out waiting for packet header" 
code is running if i look at serial monitor (but it has no littlefs files)

made a littlefs bin adress 0x200000 button on webpage works ok on nodemcu esp12E and wemos uno sized 12F

but does not work for wemos D1 mini ESP12F

cannot figure out yet what is the problem

ldijkman, 2022-05-09:
https://esphome.github.io/esp-web-tools/
tryed the install 
worked on my amica nodemcu ESP8266 / ESP12E (nice the improv configure wifi ! )
wemos d1 mini ESP12F also worked

ldijkman, 2022-05-10:
changed the baudrate in local js file to 57600 now it installs littlefs bin at 0x200000 on wemos d1 mini esp12F

https://github.com/ldijkman/randomnerd_esp32_wifi_manager/issues/28

https://ldijkman.github.io/randomnerd_esp32_wifi_manager/Touch_Electra-Electra_Touch/local_js.html

degraafm76, 2022-09-27:
with the latest version flashing ("Amica") NodeMCU with CP2101 is working

closed 2022-09-27 by degraafm76

rbric, 2023-06-20:
Same issue... I moved the VIN pin to the RST pin. It connected and flashed. In case someone is stuck also