esphome/esp-web-tools #160: Failure when attempting to flash specific firmware

thorrak, 2022-01-16:
I'm currently trying to set up ESP Web Tools to flash the firmware for one of my projects and keep getting a "Timed out waiting for packet header" error midway through flashing. I've now attempted on several different ESP32 boards, and all encounter the error at roughly the same place. By contrast, using the "live demo" at https://esphome.github.io/esp-web-tools/ does not result in an error - the firmware flashes successfully.

Manifest: https://www.brewflasher.com/firmware/api/firmware_manifest/55/

Log:
```
Try hard reset.
connect-495e0d76.js?23f6:314 Chip type ESP32
connect-495e0d76.js?23f6:314 Uploading stub...
connect-495e0d76.js?23f6:314 Running stub...
connect-495e0d76.js?23f6:314 Stub is now running...
connect-495e0d76.js?23f6:314 Detecting Flash Size
connect-495e0d76.js?23f6:314 Flash Manufacturer: ef
connect-495e0d76.js?23f6:314 Flash Device: 4018
connect-495e0d76.js?23f6:314 Auto-detected Flash size: 16MB
connect-495e0d76.js?23f6:314 Writing data with filesize: 17056. Compressed Size: 11160
connect-495e0d76.js?23f6:314 Took 995ms to write 11160 bytes
connect-495e0d76.js?23f6:314 Erase size 0, blocks 0, block size 16384, offset 0x0000, encrypted no
connect-495e0d76.js?23f6:314 Writing data with filesize: 2031616. Compressed Size: 102808
connect-495e0d76.js?23f6:314 Took 10144ms to write 102808 bytes
connect-495e0d76.js?23f6:314 Erase size 0, blocks 0, block size 16384, offset 0x0000, encrypted no
```

TD-er, 2022-01-16:
What size of bin file do you try to flash?
What flash frequency have you set when compiling your build?
Are you trying to flash several files (bootloader, bin, etc) or is it a single file?
Do you also use a file system and is it included in the bin file you try to flash?

thorrak, 2022-01-16:
I'm trying to flash five separate bins using that manifest:

- **bootloader**: 17kb
- **partitions**: 3kb
- **otadata**: 8kb
- **firmware itself**: 1.4Mb
- **filesystem (SPIFFS)**: 2.0Mb

The bootloader (and the firmware) are all set to use 40Mhz flash frequency

thorrak, 2022-01-16:
Interestingly, I just re-uploaded each of the bins, and it now appears to be working, so I'm thinking this might have been human error on my part. Going to try with one or two other firmware then this may be good to close out.

thorrak, 2022-01-16:
Yep - Seems to be working now. Sorry about that!

closed 2022-01-16 by thorrak