esphome/esp-web-tools #179: Setting "new_install_prompt_erase" to true not working as expected for me

witnessmenow, 2022-02-16:
Judging by #102 , setting `new_install_prompt_erase` to true, should allow the user the option of skipping erasing the flash, but that is not happening for me. I must be doing something wrong, but I'm not sure what.

Using this Manifest:
https://github.com/witnessmenow/SmarterEveryDayMatrix/blob/main/Webflash/WebVersion/bins/manifest.json

When I click install I get brought straight to this page, with no option to not erase the flash

[image]

Thanks

balloob, 2022-03-08:
`new_install_prompt_erase` has to be set at a manifest level, not at a build level.

See docs at https://esphome.github.io/esp-web-tools/#manifest

closed 2022-03-08 by balloob

wilson3682, 2023-06-08:
use this instead, it is working for me:
"new_install_skip_erase": false,

as