esphome/esp-web-tools #185: Javascriipt error in case of integration to an individual website

redakker, 2022-03-03:
Hi,

I would like to implement my own uploader so I've created a small website:
`<html>
    <head>
        <script type="module" src="https://unpkg.com/esp-web-tools@7.1.0/dist/web/install-button.js?module"></script>
    </head>
    <body>
        <esp-web-install-button manifest="manifest.json"></esp-web-install-button>
    </body>
</html>
`

I have the manifest file:
`{
  "name": "MyAPP",
  "version": "1.0",
  "builds": [
    {
      "chipFamily": "ESP32",
      "parts": [
        { "path": "1.0.bin", "offset": 0 }
      ]
    }
  ]
}
`

Connect button appears, and I can see my ESP32 board, but when I choose the board I get an error:

[image]

Do you know what could be the problem and how can I fix it?
I use Chrome  98.0.4758.102

Thanks!

redakker, 2022-03-03:
Manifest file is downloaded with no problem

[image]

balloob, 2022-03-08:
Unable to reproduce this.

In Chrome dev console -> sources, enable "Pause on excecptions" (pause button on a stop board) and run it again. It will now break when the error occurs and you can inspect the manifest variable to see what was loaded.

[image]

The network tab will also show what was fetched, make sure the "response" matches the data that you expect.

redakker, 2022-03-08:
@balloob 
I cannot reproduce anymore. Mabe because of the recent code changes.
Now it is working.
Thank you for your answer!
You can close the ticket

closed 2022-03-08 by balloob