esphome/esp-web-tools #193: Allowing device filter in the manifest for request port

abishekmuthian, 2022-03-10:
Right now, The request port dialog is like this on Linux:

[Current esp-web-tools request port dialog gif]
[image]

If the device filters are accepted i.e. with `usbVendorId` & `usbProductId` :
```
            const filters = [{usbVendorId: 0x1a86, usbProductId: 0x7523}];
            const port = await navigator.serial.requestPort({filters});
```

Request port dialog can be like this on Linux:

[esp-web-tools request port dialog with device filter gif]
[image]

This setting can be useful for developers who support specific models of boards as device/vendor id doesn't change for a particular model and can reduce confusion for the end user.

balloob, 2022-03-11:
Sounds good. Pull request welcome 👍

abishekmuthian, 2022-03-12:
Sure, I'm thinking about accepting another key `filters` as it has nothing to do with `builds`?

```
"filters": [
      {
        "parts": [
          { "vendor": 0x1a86, "product": 0x7523},
          ...
        ]
      }
],
"builds": []
```

balloob, 2022-03-12:
I would just call it `requestPortOptions` and pass that object as-is when calling requestPort.

Note, we don't currently have access to the manifest when connecting.

abishekmuthian, 2022-03-14:
>I would just call it requestPortOptions and pass that object as-is when calling requestPort.

Upon further checking, it needs to be `SerialPortFilter` according to the w3c-web-serial type definitions (https://github.com/DefinitelyTyped/DefinitelyTyped/blob/dc7cac7a165e9fe4d11c8db5db4eccbe41790e57/types/w3c-web-serial/index.d.ts#L76) being used in the project .

>Note, we don't currently have access to the manifest when connecting.

Noted, Would review how it's being done during flashing and would try to implement the same during connect.

balloob, 2022-03-14:
Your type link references just an object with specific keys, it's not a specific class.

abishekmuthian, 2022-03-14:
I meant we'd stick with naming conventions of the web-serial package and pass  `SerialPortFilter` in `SerialPortRequestOptions` object from the manifest?

abishekmuthian, 2022-03-15:
https://github.com/esphome/esp-web-tools/pull/195