esphome/esp-web-tools #196: WOW this is nice

ldijkman, 2022-03-19:
no issue == maybe add a discussions tab in settings?

but just want to say this is cool, very good, super
treid it on the espeasy website with raspberry pi chromium ans it worked

Electra needs this ;-)
see if i can get this working or Electra
https://github.com/ldijkman/randomnerd_esp32_wifi_manager

ldijkman, 2022-03-19:
allmost working

getting some timeout messages while programming

https://ldijkman.github.io/randomnerd_esp32_wifi_manager/ProgramESP/programesp.html

ldijkman, 2022-03-19:
Like it 
But
Cannot get it working

Binary files are ok when i use it like next
https://github.com/ldijkman/randomnerd_esp32_wifi_manager/tree/main/Binary_4_ESP8266_4MB

ldijkman, 2022-03-20:
looks like i have the 2 bin part version working
when i use second littlefs bin address part in decimal [Hexadecimal 0x200000 = 2097152]
https://ldijkman.github.io/randomnerd_esp32_wifi_manager/ProgramESP/index.html

but why is the combined bin not working on address 0

balloob, 2022-03-20:
Please refrain from using GitHub issues like a forum/chat room.

There must be an issue in how you create your merged bin. This is not an issue with ESP Web Tools. Have a look at how ESPHome uses esptool `merge_bin` command to create a single firmware: https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/post_build.py.script#L14

closed 2022-03-20 by balloob

ldijkman, 2022-03-20:
i created the combined bin with a tool from espressif
https://www.espressif.com/en/support/download/other-tools
https://www.espressif.com/sites/default/files/tools/flash_download_tool_3.9.2.zip

so i would say thats ok

but ik wil have a look at your link
https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/post_build.py.script#L14

ldijkman, 2022-03-20:
tryed merge with esptool 
but again stopped programming at +/-45% https://ldijkman.github.io/randomnerd_esp32_wifi_manager/ProgramESP/programesp.html

/home/pi/Downloads/arduino-esp32-master/tools/esptool.py --chip esp8266 merge_bin -o merged-flash.bin --flash_mode dio --flash_size 4MB 0x0 ESP8266-LittleFS-wifi-manager_with_editor.ino.bin 0x200000 ESP8266-LittleFS-wifi-manager_with_editor.mklittlefs.bin

ldijkman, 2022-03-20:
esp web tools does not do my combined bin or larger filesystem littlefs

but esp web flasher does do it

ESP Web Flasher loaded.
Connecting...
Connected successfully.
Try soft reset.
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Timed out after 100 milliseconds
Connected to ESP8266EX
MAC Address: 40:F5:20:3F:88:07
Uploading stub...
Running stub...
Stub is now running...
Writing data with filesize:4169728
Erase size 4169728, blocks 8144, block size 512, offset 0x0000, encrypted no
Took 444985ms to write 4169728 bytes
To run the new firmware, please reset your device.