esphome/esp-web-tools #203: S3 flashing does not work with esp-web-tools

Jason2866, 2022-03-24:
Flashing the S3 via ESP-WEB-Flasher does work fine.

When using Esp-web-tools it does not work. I have no idea what is going wrong.

Repo with test setup: [https://github.com/Jason2866/esp-web-tools](https://github.com/Jason2866/esp-web-tools) and the flash page [https://jason2866.github.io/esp-web-tools/](https://jason2866.github.io/esp-web-tools/)

balloob, 2022-03-24:
Can you elaborate "does not work"? Screenshots? Browser console logs?

Jason2866, 2022-03-24:
https://jason2866.github.io/esp-web-tools/

Seems init of device fails.

Jason2866, 2022-03-24:
Chip S3 is detected correctly in Webflasher(Chip Name and Family).
I have no idea why this is not working when called from webtools. All other MCU types do work!

balloob, 2022-03-24:
We need to update https://github.com/esphome/esp-web-tools/blob/main/src/util/chip-family-name.ts

Jason2866, 2022-03-24:
Did [this change already](https://github.com/Jason2866/esp-web-tools/blob/main/src/util/chip-family-name.ts), does not solve.

balloob, 2022-03-25:
That error is shown when:
1. We get the chip family name in ESP Web Tools format ([source](https://github.com/esphome/esp-web-tools/blob/main/src/flash.ts#L63))
2. We find a build in the manifest that matches the ESP Web Tools chip family name ([source](https://github.com/esphome/esp-web-tools/blob/main/src/flash.ts#L88))
3. If build not found, show not supported error ([source](https://github.com/esphome/esp-web-tools/blob/main/src/flash.ts#L96))

balloob, 2022-03-25:
Do you use `ESP32-S3` as your chip family in your manifest.json? hmm it looks like you do https://github.com/Jason2866/esp-web-tools/blob/main/manifest/unofficial.tasmota32.manifest.json#L8

balloob, 2022-03-25:
It should log the found chip family to your browser console

Jason2866, 2022-03-25:
[image]
[image]

balloob, 2022-03-25:
Checked and everything seems fine:

- [Your manifest has an `ESP32-S3` build](https://jason2866.github.io/esp-web-tools/manifest/unofficial.tasmota32.manifest.json)
- [Your ESP Web Tools build](https://jason2866.github.io/esp-web-tools/dist/web/connect-8b802612.js) has support for S3 in `getChipFamilyName`
- The console log shows that it's detecting it as `ESP32-S3`

btw, can you ping me on Discord? Might be faster to communicate to figure it out. I'm `balloob#6041`

balloob, 2022-03-25:
Can you make sure that in the network tab the "Disable cache" checkbox is checked and you're not looking at an old version of the manifest?

Jason2866, 2022-03-25:
I reworked web-flasher https://github.com/NabuCasa/esp-web-flasher/pull/108
Now it works :-)
Did a local build. You can try under https://jason2866.github.io/esp-web-tools/

balloob, 2022-03-25:
I can't try yet because I don't have an S3. Just ordered one though! 

Fix released in https://github.com/esphome/esp-web-tools/releases/tag/7.3.1

Let me know if it works.

Jason2866, 2022-03-25:
Going crazy....  Using the released version, i get the same error. When i put all packages local (in the repo) it does work.
The used versions are same.

Jason2866, 2022-03-25:
Tooo late, **works!!** I forgot to update in index.html....
Thx for your help, and this great tool

But now really bedtime

closed 2022-03-25 by Jason2866