esphome/esp-web-tools #218: funding_url not working.

sblantipodi, 2022-04-12:
Hi,
thank you for all the work you are doing on this winderful tool.

I'm using the latest release of esp-web-tools v8.0.0.

I have added a `funding_url` option to my JSON but it doesn't show the "fund development" button.

This is where the button should be visible:
https://sblantipodi.github.io/glow_worm_luciferin/beta.html

this is the JSON I'm using:
https://sblantipodi.github.io/glow_worm_luciferin/static/firmware_build_beta/manifest_full.json

and this is the popup without the fund development button.
[image]

where I'm wrong?

Thank you!

balloob, 2022-04-12:
The funding URL only shows up if the installed firmware is the same as the firmware in the manifest by comparing name in manifest with name of firmware that Improv returns.

Based on your screenshot the firmware reports FIRM as the name while the firmware name according to the manifest is "Glow Worm Luciferin Full".

You can override the function that determines if it's the same firmware using overrides in [ESP Web Tools 8.0](https://github.com/esphome/esp-web-tools/releases/tag/8.0.0)

closed 2022-04-12 by balloob

sblantipodi, 2022-04-13:
@balloob I'm using a different name because I want to leave the user the possibility to erase the firmware even if it's the same firmware version.
I'm using this option:
  "new_install_prompt_erase": true
but this options does not work if there is the same name and version.

is there a 
"install_prompt_erase": true
option ?

balloob, 2022-04-13:
That feature already exists. If the firmware + version are the same, an option to "Erase User Data" is offered, which erases data + installs firmware again.

[image]

balloob, 2022-04-13:
What would be the use case to ask a user to erase the data on an upgrade ? You either want to erase it or not ?

sblantipodi, 2022-04-13:
@balloob  what is the option to give the user the possibility to always clear user data?
I have same firmware, same version, but I don't see the "erase user data" button.

[image]

every flasher have the option to clear user data, it's like a "start from scratch" option and I think that many users wants it.
I have users asking me why I removed the "Start from scratch" button :)

Why don't you leave this option available for us using the esp web tools and let us choose if we want it for our firmware?

Searching the internet I have seen that you already discussed this problem here #130 
but from what I see, the "reset device" button that I can use in the "log and console" reboots the device, 
does not wipe user sata as other users requested in that issue.

what am I missing out?

balloob, 2022-04-13:
Like I wrote above, the firmware + version need to be the same. In your screenshot the version is not the same because it shows "UPDATE FIRM"

sblantipodi, 2022-04-13:
thank you again for your time.

I sincerely don't get the point why a "Erase User Data" button should be related to the "firmware name + version".
I can understand the firmware name, but why I can't erase the device if I'm running a different version?

is there a workaround to make that button always visible?

sblantipodi, 2022-04-13:
@balloob suppose that a user wants to upgrade to a new version available and wants to start from scratch with a new configuration. 
isn't this reasonable?

the old version of esp web tools offered this option:
[image]

why the new one not?

[image]

balloob, 2022-04-14:
Just erasing a device will put it in an unbootable state. That's not the goal of ESP Web Tools. 

The "Erase User Data" option executes an erase + install firmware. That's why it can't be done any other case.

sblantipodi, 2022-04-15:
@balloob I still don't get it, why don't leave the possibility to always use this option like before.
[image]

it could be always available when flashing the new firmware, no matter if same version or name.
doesn't this have sense?

sblantipodi, 2022-04-21:
Ok, thank you for all the answers and the time spent helping me here. 
I get it working as I would like. I had some problems in the improv part and that's why I was misunderstanding your explanations.

Thank you very much. Love this tool!