esphome/esp-web-tools #228: How to count the number of succesfull installs

ldijkman, 2022-05-08:
Hello

do you know how  to count the number of succesfull installs

would like to know how much it is used succesfull

https://ldijkman.github.io/randomnerd_esp32_wifi_manager/Touch_Electra-Electra_Touch/

ldijkman, 2022-05-08:
stays in my mind

i think the end "succes image" is not showed
would it be possible to choose an end "succes image" at the end
an choose image location in json file or button code

ldijkman, 2022-05-14:
[image]

balloob, 2022-06-10:
I don't want to allow customizing the ESP Web Tools UI. Instead you should implement Improv, it will allow you to let the user navigate to a website once connected to wifi. There you can show anything you want.

closed 2022-06-10 by balloob