esphome/esp-web-tools #245: 'Timed out waiting for packet header' when writing 16MB image

lexologe, 2022-06-03:
Hi, I am trying to write an image to my 16MB flash chip. I tried to use separate files like

```
{
  "name": "Latest Release",
  "new_install_prompt_erase": false,
  "builds": [
    {
      "chipFamily": "ESP32",
      "parts": [
        { "path": "bootloader.bin", "offset": 4096 },
        { "path": "partitions.bin", "offset": 32768 },
        { "path": "ota_data_initial.bin", "offset": 57344 },
        { "path": "firmware.bin", "offset": 65536 },
        { "path": "spiffs.bin", "offset": 4259840 }
      ]
    }
  ]
}
```

or a single 16MB image file like 

```
{
  "name": "Latest Release",
  "new_install_prompt_erase": false,
  "builds": [
    {
      "chipFamily": "ESP32",
      "parts": [
        { "path": "big-file.bin", "offset": 0 },
      ]
    }
  ]
}
```

but every try ends in error "Timed out waiting for packet header".

I tracked it down to the last of the separated files. The `spiffs.bin` is about 12 MB in size and may be this is related to a time out that is not long enough.

ldijkman, 2022-06-03:
Lower baudrate helped for me 
Now use 57600
But searching for how to change back to 115200 after programming

TD-er, 2022-06-03:
Not sure if related, but I noticed that the esptool.py tool also has some issues when handling full 16M files. Reading full flash always failed with 16M at once and writing sometimes fails with such large files.
So maybe there is something fundamentally different when crossing some boundary (e.g. 8M).
Still increasing some timeout may fix it, but not sure if it always does.
Therefore maybe it is best to offer the flashing into several chunks like 4M or 8M in size, since the manifest also supports this?
And maybe if your image has some chunks with lots of zeroes, it might even speed up the flashing process if you skip those sections?

balloob, 2022-06-03:
Default timeout is 3 seconds per block btw.

ldijkman, 2022-06-05:
Hello

today i treid ESP32 wemos d1 r32 32mbit/4mbyte 
some code and 2mb littlefs 
and i had to lower the baudrate again to 57600 to get rid of time outs
on a lower baudrate it is almost 100% succesfull install

https://ldijkman.github.io/randomnerd_esp32_wifi_manager/Touch_Electra-Electra_Touch/ESP32/esp-web-tools-8-0-3/Burn_the_Bits.html

lexologe, 2022-06-07:
> Not sure if related, but I noticed that the esptool.py tool also has some issues when handling full 16M files. Reading full flash always failed with 16M at once and writing sometimes fails with such large files.

I didn't test with esptool.py writing the full 16MB. It works successfully with the separated set of files.

lexologe, 2022-06-07:
> You can play with baudrates here: https://nabucasa.github.io/esp-web-flasher/ . That is the library used by ESP Web Tools.

I need to check this in esp-web-flasher but I also found a baudrate initialization in connect.ts around line 24:

```
await port.open({ baudRate: 115200 });
```

and changed it to 57600 as mentioned from @ldijkman

ldijkman, 2022-06-07:
I changed the baudrate in the connect js file
There are 2x 115200 in latest version of esp web tools
The js files i use

https://github.com/ldijkman/randomnerd_esp32_wifi_manager/tree/main/docs/Touch_Electra-Electra_Touch/ESP32/esp-web-tools-8-0-3

balloob, 2022-06-10:
Fixed in 8.0.4

closed 2022-06-10 by balloob

lexologe, 2022-06-28:
> Fixed in ESP Web Tools 8.0.4

I tried it with v8.0.6 with no luck. Even if I used separate files or a big 16mb file.

```
Try hard reset.
Chip type ESP32
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x184020
Flash Manufacturer: 20
Flash Device: 4018
Auto-detected Flash size: 16MB
Image header, Magic=0xFF, FlashMode=0xFF, FlashSizeFreq=0xFF
Writing data with filesize: 16777216. Compressed Size: 857265
Took 139506ms to write 857265 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
```

PR #248 (2022-06-10): ESP Web Flasher 5.1.4
https://github.com/NabuCasa/esp-web-flasher/releases/tag/5.1.4

* Increased timeout to 16MB spiffs flash problem ([#127](https://github.com/NabuCasa/esp-web-flasher/pull/127)) @robdobsn

Fixes #245