esphome/esp-web-tools #254: Detect fake flash

DTTerastar, 2022-06-16:
Users often report issues w/ certain devices not working correctly and blame the firmware. Could we when detecting flash try and read towards the end of flash and actually verify that the flash isn't a 2mb chip reporting it's a 4mb chip? It would help immensely on the amount of support. Those cruddy chips often report "success" when writing, but error on reading past 2mb.

balloob, 2022-06-16:
How does this show up in the logs? Do we get a write failure or just a timeout when writing above a certain %? 

Right now we show a generic error message when installation fails. That could be expand to launch some "diagnostics" mode to analyze the chip or it could redirect to a website to help with further troubleshooting (and it would analyze the chip).

TD-er, 2022-06-16:
I think these fake units will then "roll over" on the address, so when reading from the "4M-1" address, you're actually reading from the "2M-1" address (or "1M-1")
So be careful when drawing conclusions based on whether you can read from some address.

DTTerastar, 2022-06-16:
esptool.py read_flash can't read past 2mb.  I'm not saying ALL fake flash work this way, but my devkit c3 certainly does.

DTTerastar, 2022-06-16:
Yeah keep it in the console for people that have flashed something 15 times and are currently looking for something :)

Jason2866, 2022-06-22:
The problem are not fake flash chips. There are C3 devkits with 2M. When the are sold as 4MB devices or the user did not noticed that it is a 2MB device...

DTTerastar, 2022-06-22:
I can write all the way to 4MB.  I can't read past 2MB.  That's all I know atm.