esphome/esp-web-tools #260: ESP32 bootloop when flashed with esp-web-tools

tigertop, 2022-06-27:
Hi,

I am having problem with uploading firmware to ESP32 device via esp-web-tools.
I have found similar issues #151 and [here in esp-web-flasher](https://github.com/NabuCasa/esp-web-flasher/issues/83)

So after building the firmware with esphome I am able to flash the files successfully with esptool.exe
`.\esptool.exe --port COM3 --chip esp32 --before default_reset --after hard_reset --baud 921600 write_flash  -fm dio -fs detect 0x1000 firmware\bootloader_dout_40m.bin 0x8000 firmware\partitions.bin 0xe000 firmware\boot_app0.bin 0x00010000 firmware\firmware.bin`

### Manifest
```
{
  "name": "Test",
  "version": "1.1",
  "builds": [
    {
      "chipFamily": "ESP32",
      "parts": [
        {
          "path": "firmware/bootloader_dout_40m.bin",
          "offset": 4096
        },
        {
          "path": "firmware/partitions.bin",
          "offset": 32768
        },
        {
          "path": "firmware/boot_app0.bin",
          "offset": 57344
        },
        {
          "path": "firmware/firmware.bin",
          "offset": 65536
        }

      ]
    }
  ]
}
```

But when esp-web-tools are used (tested V7.1.0 and V8.0.1) the endless reset loop is shown in terminal log:
```
rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jul 29 2019 12:21:46
```

I have also teste with `dio` mode. I am setting the mode from the ESPHome YAML:
```
esphome:
  name: XXXXXXX
  build_path: ./build/
  name_add_mac_suffix: true
  project:
    name: "XXXX.XXXXX"
    version: "1.0.8"
  platformio_options:
    upload_speed: 460800
    monitor_speed: 115200
    board_build.partitions: ../XXXXX_partitions.csv
    board_build.flash_mode: dout
```
### Firmware verification
I have also tried to verify the written firmware with esptools.exe. When written via esptools.exe from console all parts are normally verified, but when written via esp-web-tools I get the following mismatch error:
```
esptool --port COM3 --chip esp32 --before default_reset --after hard_reset --baud 921600 verify_flash  -fm dio -fs detect 0x1000 firmware\bootloader_dout_40m.bin
```

Error
```
esptool.py v3.3.1
Serial port COM3
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:94:54:33:4f:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash params set to 0x0230
Verifying 0x42d0 (17104) bytes @ 0x00001000 in flash against firmware\bootloader_dout_40m.bin...
-- verify FAILED (digest mismatch)

A fatal error occurred: Verify failed.
```

### esp-web-flasher
I have also tried uploading with the [esp-web-flasher](https://github.com/NabuCasa/esp-web-flasher/issues/83) and I am also getting the bootloop situation.

I also tried to flash via esp-web-tools and then flash only the bootloader part with esptool.exe and the firmware works that way.
And an another unsuccessful test was to upload all parts via console (works) and then flash all parts except for the bootloader via esp-web-tools.

I would really appreciate your help on this issue as I am stuck on this for a around a week.

Thanks.

mkrasselt1, 2022-07-02:
I am also trying to flash larger files, but narroed it down to beeing unable to flash even the bootloader 
`{
    "name": "MySoftware Software",
    "version": "0.1.1",
    "new_install_prompt_erase": true,
    "builds": [
        {
            "chipFamily": "ESP32",
            "improv": false,
            "parts": [
                { "path": "/bootloader_dio_40m.bin", "offset": 4096 }
            ],
            "unused":[
                { "path": "/bootloader_dio_40m.bin", "offset": 4096 },
                { "path": "/partitions.bin", "offset": 32768 },
                { "path": "/boot_app0.bin", "offset": 57344 },
                
                {
                    "path": "/firmware_0.1.1.bin",
                    "offset": 65536
                }                
            ]
        }
    ]
}`

I flashed the device with pio first, tryed flashing only the app throuh webtools (working fine) and then trying to flash only the bootloader and having it fail to a bootloop, not beeing able to open the console as its beeing flooded with data

TD-er, 2022-07-02:
Do you know with what version of esptool and elf2bin the binary was created?

The newer version (4.x) does call elf2bin with parameters to append an SHA checksum of the binary.
When directly flashing using esptool, it will detect the flash speed and size and patch the binary when flashing.
But if it was already "signed" with such a checksum, it will not patch the binary when flashing to set the correct flash size and flash clock frequency.
Simply because changing these bits will corrupt the checksum.

The 4.x version of esptool does check for this, but if the bin was made including the SHA and esptool doesn't check for it, it may corrupt the bin while flashing.
On the other hand, if the wrong flash parameters were set in the board definition or PIO env, then esptool cannot correct for it when flashing.

So you could generate a merged binary including the bootloader using esptool and try to upload that using the web flasher.
Just make sure to have the flash frequency and size set correctly as the web flasher tool does not patch the binary while flashing.

Jason2866, 2022-08-15:
@brentru  Sounds strang but this mismatch is a good sign? It does mean the needed header patching is done to set the correct flash mode, flash frequ and flash size.
So the SHA has become invalid (which was added during compiled and altered from esptool before flashing).

amandel, 2022-10-01:
Might be this helps others, I solved this by creating a dedicated boot loader with the correct parameters for my esp32 using the command:
```bash
esptool.py --chip esp32 merge_bin --output bootloader.bin  \
    --flash_freq keep --flash_mode dio --flash_size 4MB \
    --target-offset 0x01000 \
    0x01000 bootloader_dio_40m.bin
```
and then using `bootloder.bin` to flash the esp.

Testing the result via `esptool.py --chip esp32 image_info --version 2 bootloader.bin` shows if it worked.

Jason2866, 2022-10-01:
Yes a way to do it. It can be done more easier. Latest Arduino bootloader files do NOT have SHA checksums. So using the needed one from the already precompiled is working too. For creating a valid all in one firmware, esptool.py can be used using the `merge` command and setting the wanted flash size, flash frequency and flash mode to generate this file.