esphome/esp-web-tools #270: Release script doesn't work with 9.0.1

balloob, 2022-07-20:
Logs: https://github.com/esphome/esp-web-tools/runs/7423186279?check_suite_focus=true#step:5:16
Output misses necessary files: https://unpkg.com/browse/esp-web-tools@9.0.0/dist/web/

Error claims it can't find `esploader-js` during bundling.

Unable to reproduce locally.

ithinkido, 2022-07-24:
I can reproduce this . Run Localy on v 9.0.3
```
~/Desktop/esp-web-tools-9.0.3$ ./script/build

dist/install-button.js → dist/web...
(!) Unresolved dependencies
https://rollupjs.org/guide/en/#warning-treating-module-as-external-dependency
esptool-js/esploader.js (imported by dist/flash.js)
created dist/web in 4s
```  

I am also no able to upload code from the Demo page [here](https://esphome.github.io/esp-web-tools/). The install hangs on "Preparing instalintion"   

Tested on ESP8266 and ESP32 boards

with the following console log errors 
[image]

amandel, 2022-08-28:
I came across this. Looks like the little PR should get this fixed.

closed 2022-08-28 by balloob