esphome/esp-web-tools #273: Serial port is not ready

tobiyas09, 2022-07-27:
Hi,
I used ESP-Prog to connect the esp32 to the PC and when I click on the connect button the first time, this error pops out.

[image]

Then when I click again in the port dropdown menu it says (Paired), there is no error and I can use the tool.
Any idea why? I guess the error pops out because of the pairing. It can be annoying because I need to click on connect button two times.

alentz44, 2022-08-05:
I'm having the same issue. Please let me know if you find a work around. Thanks!

balloob, 2022-08-05:
Can you both please include the OS/Browser and exact device you're trying this with?

That error is shown when Improv is unable to open the connection to the serial port, for which it waits a second ([src](https://github.com/improv-wifi/sdk-serial-js/blob/main/src/serial.ts#L72)).

tobiyas09, 2022-08-08:
I'm using ESP32-WROVER-IE-N8R8, Ubuntu 20.04, Google Chrome v103.0.5060.134

balloob, 2022-08-08:
How do you run Google Chrome, inside a Snap or directly?

tobiyas09, 2022-08-08:
> How do you run Google Chrome, inside a Snap or directly?

Directly.

balloob, 2022-08-08:
Can you reproduce this with a different OS/Browser/Device. I am not seeing it on my end.

alentz44, 2022-08-13:
@tobiyas09, if it helps I instead just followed these instructions to flash directly.

https://www.instructables.com/Flash-WLED-to-ESP8266-Based-Controllers/?amp_page=true

Jason2866, 2022-08-15:
Never tried to use ESP-Prog with esp-web-tools. It is uncommon since it is expensive in relationship to "normal" usb-serial converters (flashers).
For reliable flash i can recommend the "golden" ch340 board. It has a 3.3V LDO chip which does power the esp chip reliable when flashing.

BroManDudeGuyPhD, 2023-01-24:
I also ran into this issue with both Chrom and Edge

SymlincJoe, 2023-11-28:
This is older now, but I am having the same issue; Windows with ESP32 board

gcormier, 2023-11-30:
Same issue on a Linux laptop and CP2104N tty converter.

Command line works fine with esptool.py.

[image]

```
greg@grexps:~$ esptool.py chip_id
esptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:f0:9e:ce:c1:98
Stub is already running. No upload is necessary.
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: c8:f0:9e:ce:c1:98
Hard resetting via RTS pin...
```

Aarushraj-Puduchery, 2024-08-27:
connect the board, run the EEPROM Clear code, unplug it, plug it in again and try it. it worked for me.

embediver, 2024-12-09:
I know this is quite an old issue, but I am also experiencing this error.

I am using Chromium `Version 131.0.6778.85 (Official Build) Fedora Project (64-bit)` and trying to flash Tasmota using their [Web-Installer](https://tasmota.github.io/install/).

I got the first device flashed, but also got the "_not ready_" which I somehow fixed by reloading the site and replugging the FTDI I have.

For the second device I couldn't get it to work.
The FTDI shows correctly in Chromium (with `Paired` annotated) but the error persists.
Reconnecting and restarting Chromium didn't change anything.
Using _esptool.py_ works without any problems (didn't reconnect anything, runs without sudo, ...).