esphome/esp-web-tools #277: [Feature] allow serial speed to be increased

amandel, 2022-08-08:
If I flash via command line I use upload_speed of 921600. It would be great if this is also possible via esp-web-tools, might be as a additional parameter (per `build`) in the manifest?

balloob, 2022-08-08:
The reason I haven't made it configurable yet until now is because switching the baud rate on a serial port that has been used before on the old baud rate is causing the serial port in the browser to stop working (and I haven't spend time to investigate why). Ideally we would go to a faster speed when we know it is supported.

The current speed is chosen because it works for both good and shitty UART chips.

Allowing you to configure one means it might not work on every chip. Now as a firmware owner you probably know what you do and don't support so this can be considered.

TD-er, 2022-08-08:
> Now as a firmware owner you probably know what you do and don't support

Wow, tricky assumption ;)

But on a more serious note, can you see what the current baudrate is?

balloob, 2022-08-08:
115200. It's currently defined in [3 different places](https://github.com/esphome/esp-web-tools/search?q=115200)

TD-er, 2022-08-08:
I meant checking the serial port to ask its set baudrate.
Not the speed you (try to) set it to.

amandel, 2022-08-08:
Thanks for the details. Actually we only use one devkit, but still there are many variants and clones out there.
I also like a no options but working solution 😃 so I see your points. But we have groups that install several devices in a row. There it would specially help to speed it up. 
I'm not too deep into this, would it be sufficient to adjust the 3 places you mentioned with a higher speed? If so, I could give it a try and feedback (will take some time)

balloob, 2022-08-08:
If it makes your life easier, you could open a PR first to extract it into a single constant. That would make it configurable later on easier too.

Jason2866, 2022-08-15:
> If I flash via command line I use upload_speed of 921600. It would be great if this is also possible via esp-web-tools, might be as a additional parameter (per `build`) in the manifest?

Flashed a lot of different devices (beeing a member of Tasmota). Tried to increase the speed from 115200 to higher ones. Finally i went back to 115200 as standard for Tasmota setup. Even devices from the same type (which one worked with higher serial speed) are not safe to use with higher speeds.
You never know what clone USB serial chip is used.

amandel, 2022-08-28:
I had good results using my test device (which does say nothing for the field, I know). Seeing the increase in speed, it might be worth to give it a try and fallback if it does not work. I've to say again that I only work with "esp32dev" boards having to flash _any_ device you are likely better of not putting it in a strange state where you might loose communication. 

I had do do minimal changes to the call to the `esploader.js` where I'm not sure why the initial [`change_baud`](https://github.com/espressif/esptool-js/blob/0c1b972a05d691c85da23fcc937d91dcf7e283eb/ESPLoader.js#L652) call as part of [`main_fn`](https://github.com/espressif/esptool-js/blob/0c1b972a05d691c85da23fcc937d91dcf7e283eb/ESPLoader.js#L667) does not work. Might be `esploader.js` needs some more love or I've to dive deeper into JS.

The change I did is applied as "patch" to `esp-web-tools` in this [Dockerfile](https://github.com/openbikesensor/OpenBikeSensorFlasher/blob/feature/flash-with-921600/Dockerfile), see the changes here: https://github.com/openbikesensor/OpenBikeSensorFlasher/pull/25/files