esphome/esp-web-tools #279: [Feature Req] Option to read flash and download read data

TD-er, 2022-08-09:
Since now all features of esptool.py are implemented (or being implemented?), I would really like to see an option to image/backup the flash of an ESP.
Either the entire flash, or sections of it.

It would be extremely nice if it was possible to read the partition info and thus allow for downloading named sections like app0, the file system area, etc. with bonus points to have the esp-web-tools manifest included.

But just imaging/cloning of a device would already be a great feature to have.

balloob, 2022-08-10:
[esptool.js](https://github.com/espressif/esptool-js/) is meant to be a full featured port of esptool.py. esptool.js is one of the many libraries used by ESP Web Tools.

The goal of ESP Web Tools is still to empower firmware authors to offer an easy way to their users to install and manage devices.

What you're looking for sounds like something that can be made with esptool.js but falls out of scope for ESP Web Tools.

closed 2022-08-10 by balloob

TD-er, 2022-08-10:
OK, I get it that it may be out of (the intended) scope of the tool.
But maybe a suggested use case can persuade you to consider?
For example for helping to debug cases where flash parameters are not working and people may not be experienced enough to install and run esptool.py?
Or to backup a unit to test some other firmware?

balloob, 2022-08-10:
I think that sounds like it should be done by another tool powered by esptool.js. I don't want to add feature creep to ESP Web Tools. Every extra feature might confuse the user and end up in not ending up with running your firmware.