esphome/esp-web-tools #280: support for standard, non-proprietary browsers

petaflot, 2022-08-10:
using a recent version of firefox, I get this message

http://lutolf.vserver.nimag.net/files/wled.png

I mean... seriously? like it's compulsory to have a proprietary browser to see your website? this is bullsh*t

TD-er, 2022-08-10:
as far as I know, not all browsers support to access serial ports. Or at least not yet.

Not sure if it is already an official standard to allow access or that it may be a security setting which may need to be changed to allow this access.

But I guess that if you can find proof your browser does support this, the people behind this tool may be more than happy to remove this restrictive browser check.

petaflot, 2022-08-10:
displaying the content of the website does obviously not depend on accessing the serial port ; if one just wants to see what your project offers there is no need to force the user in such a way.

does this mean that one cannot access your browser without an ESP kit? that would be absurd. and if one has no uC unit to plug on a serial port (or, for some reason, no serial port available) they would also not be able to see your website for let's say read the doc?

@TD-er the cause you mention is not a reason, it's an excuse and a lame one too.

TD-er, 2022-08-10:
I do understand it looks like there is an early check blocking you to even try it.
But given the update frequency of this project, I can imagine the devs are well aware of which browser is and isn't capable of handling it.
So it is a design choice to block prematurely instead of failing late and maybe have a lot of reported issues from users.
Whether or not that's a bad (or lame) design choice is debatable.

petaflot, 2022-08-10:
well let's have this debate, then.

I understand that some might prefer "block all rather than accept the event of a later failure" but IMHO this is just a result of poor programming and it should at least be possible to "accept the risk of an unexpected failure and proceed"

by poor programming I mean that events should degrade gracefully, ie. if there is such an error then a short error message should be displayed and not propagate to who-knows-where

TD-er, 2022-08-10:
> well let's have this debate, then.

Not sure if I'm the one to have such a debate with, since I'm not the dev of this project.
Such design decisions are always a trade-off between managing number of reported issues/confused users and maintaining a more complex structure of possible fail options and strings displaying those.
And as always with such concessions, there's always something that's less optimal.
Handling lots of error situations may confuse users.
Bluntly stopping as you already know things won't work may upset others who might be perfectly capable of understanding specific errors.

balloob, 2022-08-10:
@petaflot Firefox has decided to not implement [WebSerial](https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API) and so the web installer cannot be used. You should request for it in their issue tracker.

You've also raised an issue at the ESP Web Tools library. We are not affiliated with the WLED installation website that you see. It's also not the WLED main website of the project that you claim blocks your access.