esphome/esp-web-tools #289: Improv WiFi works most of the time but not always.

sblantipodi, 2022-09-09:
Hi,
I have seen that improv wifi is not really reliable.

It works most of the time but not always.

Sometimes, when I try to flash the ESP8266, it asks me for the WiFi SID and password,
I enter the credentials, and then it hangs on this screen.

It happen on my implementation here:
https://sblantipodi.github.io/glow_worm_luciferin/beta.html

and on the one on the official esp web tools page.

[image]

Is this a bug? I tried changing the router and the esp8266 but same problem.

balloob, 2022-09-09:
So this can be caused by:
 - The implementation of Improv on your device
 - The device unable to reliably connect to the Wi-Fi

Can you try to use the demo firmware that ESP Web Tools website tries to install?

sblantipodi, 2022-09-10:
Hi balloob, thanks for the answer.

Sometimes it happen even with the demo here:
https://esphome.github.io/esp-web-tools/

Is there a reference implementation that shows how to correctly implement it?

thanks

balloob, 2022-09-13:
We do not currently have a reference implementation.

sblantipodi, 2022-09-14:
I confirm that the problem happens even with the demo on the official esp web tools page.

[image]

the tools hangs after entering the WiFi credential.

I'm trying to debug it using my implementation, it seems that sometimes credentials are not received by the ESP,
esp web tools wait for an answer from the ESP and ESP wait for the credentials.

is there a way to tell esp web tools to retry sending credentials if no answer is received?

balloob, 2022-09-14:
I'll look into adding a timeout to the command. Can you try it with a different device?

sle118, 2022-11-04:
@balloob I can also confirm this being an issue in my effort to support improv on the squeezelite-esp32 project, for which I am exposing an installer here: https://sle118.github.io/squeezelite-esp32-installer/

In my case, there were 2 factors to consider
- Our implementation is multi-threaded, which initially resulted in IMPROV packets overlapping with logs from other threads. This was easily fixed by using the esp-idf's uart_write_bytes, which grabs an exclusive lock on the serial port during transmission. 
- Our console i/o thread starts roughly at 4 seconds after resetting the device with a substantial amount of logs produced during that time. 
 
Being able to setup the "boot timeout" period is a critical feature, as users have to try multiple times with the current 1 sec timeout for the improv stack to be recognized. I have forced various timeout durations in my local npm modules and can confirm this makes a positive difference.

Note that as of writing this, the improv enabled code hasn't been committed yet to the squeezelite-esp32 repo; coexistence with our telnet implementation as well as our web interface, which pipes UI commands back into the console thread made this a bit more tricky to implement. It's worth noting that I have written the improv implementation in 'c' to possible release a stand-alone library for others to use.

balloob, 2022-11-18:
You can use the [`new_install_improv_wait_time`](https://esphome.github.io/esp-web-tools/#manifest) option to extend the delay on how long it will wait for the Improv response.

sle118, 2022-11-18:
This will certainly improve new installations, but I am referring to using the improv button as stand alone when not actually flashing binaires. Perhaps this is an issue I should open separately in that repo instead of here then again I'll see if that mechanism, used by the installer, could be leveraged as stand alone. 

For the record, here is the link to the work in doing
https://github.com/sle118/squeezelite-esp32/issues/187

balloob, 2022-11-18:
When you start ESP Web Tools to just use Improv, you need to be very fast to hit connect + select the device within 4 seconds from booting to have Improv not respond?

sle118, 2022-11-18:
@balloob I just pushed a possible enhancement for the ones (like me) who'd like to use improv serial as a stand-alone option, outside of the "new flash" context. This allows specifying a custom timeout like this:
`<improv-wifi-serial-launch-button id="button_improv_serial" learnMoreUrl="https://www.esphome.io/components/improv_serial.html" timeout="5000"></improv-wifi-serial-launch-button>`

let me know your thoughts

balloob, 2022-11-18:
So now I am a bit confused. We're discussing this in the ESP Web Tools repo but your example code shows the standalone Improv Wi-Fi SDK. You don't need to use that one, as the ESP Web Tools can provide installation + provisioning.

Did you see my question about that one has to be very fast to hit connect to actually send Improv within 4 seconds of a device booting up?

sle118, 2022-11-18:
For the new installation, I'll definitely go with the manifest approach, which fits my requirements. As for the standalone button, I did hijack this issue involuntarely because I had not fully checked the implementation.  I opened a new issue in the sdk-serial-js repo instead. The reason why I am using the stand-alone button is that my UI will show the "provision wifi" button until/if user actually picks a firmware to flash on the system. This is to provide a simple way to provision wifi and check logs without going the route of picking up an actual firmware from the list.

Here's the link to the serial sdk issue.

As for myself, using the manifest timeout for the esp-web-tools button will do the trick.
https://github.com/improv-wifi/sdk-serial-js/issues/95

balloob, 2022-11-18:
Improv SDK doesn't provide an option to show logs or funding URL.

I think that you've made it yourself so difficult with all the different versions per device! Why not just have a simple UI that shows the latest?

sle118, 2022-11-19:
We are only two developers and can't possibly cover all scenarios in our limited regression tests for squeezelite-esp32. In a production environment with a team ready to take issues and respond in a timely manner, I would not go that far out. But in that situation, being able to roll back to the previous release has often been the best contingency plan for our users.

The ability to play music and synchronize multiple zones, once it has become a given in a home, cannot suffer failure ever. Experience let me to realize that not being able to play music when needed can quickly become a P1 issue, and that rolling back is often the best course of action.

I've had lights not automatically turning on our off without a comment. But music playback failure is not an option here. It's mission critical ;)

balloob, 2022-11-20:
Alright, well your call. [Something like this](https://esphome.github.io/media-players/) but with a version dropdown would allow you to always use ESP Web Tools.

Will close this issue as it's Improv SDK related.

closed 2022-11-20 by balloob