esphome/esp-web-tools #291: New project using esp-web-tools

sle118, 2022-10-08:
I just wanted to say thank you.

It's been a while since @balloob suggested I use this excellent tool for the squeezelite-esp32 project, and I'm glad I did. I've modified the build workflow to automatically push artifacts and binary files when we release a new version, and the installer is hosted here
[https://sle118.github.io/squeezelite-esp32-installer/](url)

So big thank you, as I think this makes the adoption of new devices much easier for new comers. Next on my plate is to implement improv

balloob, 2022-10-10:
Awesome! I will update the site when I find some time. Great work 🎉

sle118, 2022-10-11:
now if only there was a bare-bone improv implementation targeted at the esp-idf rather than at the Arduino, that would make my life much easier for integrating it into Squeezelite-ESP32.

amandel, 2022-10-11:
Might be this one https://github.com/openbikesensor/OpenBikeSensorFirmware/blob/main/src/obsimprov.cpp / https://github.com/openbikesensor/OpenBikeSensorFirmware/blob/main/src/obsimprov.h is of help. It is also a mixture and specific to my use case but who knows.

sle118, 2022-10-11:
@amandel I appreciate the input!

Since our project (squeezelite esp32) is written for the most part in c, I've decided to bite the bullet and do the implementation myself (in bare C, using the espressif APIs). It's also worth noting that I'm also adding Bluetooth, which I suspect will be very handy when roaming around with portable speakers.

I'm curious to see how the ESP IDF will behave in dual BT (A2DP)+BLE modes.

sle118, 2022-10-12:
ok. I'll go ahead and close this issue, as it's more of a discussion than anything else. It does look like Bluetooth improv isn't going to happen for Squeezelite-ESP32, at least until (if) we decide to change the partition layout to accomodate for larger binaries. Adding the gatts piece ended up increasing the size of the generated binary by quite a lot.  So I'll concentrate on trying to add improv support from the console for now.

PR #300 (2022-12-31): Add Squeezelite-ESP32 to used projects
Add Squeezelite-ESP32 to the projects that use it. Thanks to @sle118 adding the installer to Squeezelite-ESP32.

Fixes #291
closed 2022-10-12 by sle118