esphome/esp-web-tools #292: Unable to flash T-Display-S3?

nicpottier, 2022-10-24:
Thanks again for such a great project. I've got it setup and it is such a game changer in allowing projects to get wider audiences:
  https://nicpottier.github.io/gnat/

I have web installs working great for an M5StickCPlus and the "original" ttgo (see above) but I'm not having any luck getting it to work with the new ESP32-S3 TTGO:
  http://www.lilygo.cn/prod_view.aspx?TypeId=50062&Id=1411&FId=t3:50062:3

When I try I get the usual "install" and "logs" screen, where I can indeed pull logs, but when I select INSTALL it stalls on "preparing installation", eventually timing out with an "Installation failed" message:
  `Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.`
  
Have you all had any luck flashing either an S3 based ESP32 or that specific board? Any ideas in debugging this?

nicpottier, 2022-10-24:
Possibly related?
  https://github.com/espressif/esptool-js/issues/47

balloob, 2022-10-24:
This is an upstream issue.

closed 2022-10-24 by balloob

nicpottier, 2022-10-24:
Heh, it may be caused by an upstream bug but still an open issue on this resulting library but ok. :)

nicpottier, 2022-10-24:
For future brave explorers this was a series of issues on my side:

1) need to hold down boot AND press reset to enter bootloader on this board. Not sure if this is an S3 specific thing, if so and since we can detect the board perhaps we should change the message about that?

2) I needed to use an S3 version of the bootloader. Found the files (and offsets) platform IO was using by running with verbose on, see: https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/flashing-firmware.html#id3

3) With those new binaries I had to update the offset for my bootloader to 0 in my manifest from the previous 4096 I had on original(?) ESP32 boards when using `bootloader_dio_40m.bin`

With those three changes things work fine with latest version of esb-web-tools.

balloob, 2022-10-24:
We need to talk to the bootloader to detect the type of board. I guess if the manifest contains an S3, we could change the instructions.

nicpottier, 2022-10-24:
Ya, I mean ideally we wouldn't require that at all. `esptool.py` doesn't seem to need it so I wonder what is missing on the JS version to let that happen. Anything that doesn't require special user action is good from an adoption perspective.

TD-er, 2022-10-24:
Maybe a different "reset mode"?
I know for the ESP8266 there are also different reset sequences to toggle CTS anr DTR to trigger the GPIO-0, RST, release GPIO-0.
The difference between "nodeMCU reset" and others. (forgot their names)