esphome/esp-web-tools #293: Can't add "esptool-js": invalid package version undefined

mesopa, 2022-10-30:
After many months I tried again web ESP Tools, felt very hipped because I saw Typescript but got disappointed again.

Running the following command to install the library on my existing React project:
`yarn add esp-web-tools` 

Outputs the following error message:

`
error Can't add "esptool-js": invalid package version undefined.
info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.
`

Does anyone has any hints? Something that I'm missing.
Much appreciated.

balloob, 2022-10-30:
Yarn complains about the esptool-js package not having a version number. It's a limitation of Yarn, NPM has no such problem.

I recommend to use the installation instructions on our website which recommends to include it from a CDN: https://esphome.github.io/esp-web-tools/#add-website

mesopa, 2022-11-01:
That solved my problem. Much appreciated.

Since is is a model nicely Trypescripted, could be really nice to make it work with React.

closed 2023-01-18 by balloob