esphome/esp-web-tools #297: Support request: Error writing filesystem image for ESP8266

PaulWieland, 2022-12-02:
I'm trying to setup ESP Web Tools to write a two part (firmware and file system) to an ESP8266. You can try [it here](https://paulwieland.github.io/ratgdo/flash.html) (choose v2.0a (alpha) - rolling code support, which uses this [manifest.json](https://paulwieland.github.io/ratgdo/flasher/manifest_alpha.json))

The first `ratgdo_esp8266_v2.0a.bin` file at offset `0` seems to write okay, but then it times out right after it starts to write the second `ratgdo_esp8266_v2.0a_littlefs.bin` (~3mb) file at offset `1048576`. 
[image]

I got the offset number from platform.io when uploading the filesystem:

```
Wrote 3121152 bytes (679027 compressed) at 0x00100000 in 74.9 seconds (effective 333.3 kbit/s)...
```
`0x00100000` in decimal is `1048576`

Here is the output of the browser console during the flashing process:

```
Writing at 0x48000... (100%)
flash_defl_block 8d 097
check_command write compressed data to flash after seq 18
Read with timeout 5000
Wrote 419216 bytes (296238 compressed) at 0x0 in 28.729 seconds.
Data Length 3121152
Image Length 3121152
_update_image_flash_params keep keep keep
Compressed image 
Uint8Array(679025)
Compressed 3121152 bytes to 679025...
check_command enter compressed flash mode
Read with timeout 3000
Write loop 1048576 0 42
Writing at 0x100000... (2%)
flash_defl_block 78 0da
check_command write compressed data to flash after seq 0
Read with timeout 5000
Write loop 1048576 1 42
Writing at 0x104000... (4%)
flash_defl_block d5 08b
check_command write compressed data to flash after seq 1
Read with timeout 5000
```

Does anyone know what I'm doing wrong?

hacktronics, 2022-12-19:
I am also facing the timeout issue, while writing spiffs.bin to esp32.

PaulWieland, 2022-12-19:
I wish I could get some support on this. I can't figure out how to modify the read with timeout value, or even where in the source code that it's set.

playground, 2023-03-03:
@PaulWieland did you get this resolved?

PaulWieland, 2023-03-03:
No, unfortunately not.

Jason2866, 2023-03-08:
Using latest Version? Changes are done in used esptool.js regarding timeouts.

thorrak, 2023-07-02:
What is the specific device you are having this issue with?

I ask, as I've had the latest Lolin D1 Mini v4.0.0 devices fail, where earlier versions (v3.x, with the Micro USB connectors) succeed. Interestingly, this failure occurs both with ESP Web Tools as well as with older esptool versions, but seems to succeed with the latest esptool.py.