esphome/esp-web-tools #301: Question: Possible to choose different firmware versions?

beniroquai, 2022-12-14:
First of all: Wow! This tool is amazing and will help us to distribute our firmware. Any attempts to unify it in python did not work due to OS-related issues. This is just insane what you created here! Thank you sooo much!! :-)

now, after setting everything up, I was wondering if you could also ship different versions for the ESPs e.g. version 2022.12.12, but also 2022.12.11 or alike. All for one MCU e.g. ESP32. I wasn't sure if the manifest could handle it. 

Many thanks in advance!

balloob, 2022-12-14:
That's not something we want to support natively in the ESP Web Tools UI. It's already getting quite complicated with all the different permutations.

Instead, you should make a version selector in the UI around the button and have that change the manifest attribute. For an example, see this: https://esphome.github.io/bluetooth-proxies/

closed 2022-12-14 by balloob

beniroquai, 2022-12-14:
Wow! You are amazing! Thank you so much! I'll directly implement this! :-) 

Follow-up question - almost related: Is there a way to have "post script" running on the serial, e.g. a configuration string posted via serial-write? I can definitely search for related issues in github.

balloob, 2022-12-15:
ESP Web Tools leverages Improv Wi-Fi to ask the user for wi-fi credentials. As the last step of Improv, once connected, the device can return a URL (including a self-hosted one) to the user to visit. There you can do any further onboarding as you wish.

beniroquai, 2022-12-16:
Thanks! Will dig into that! 
Thanks again for making that open source. This dramatically helps me to share our code (nobody will ever be able to compile it :D )
For the reference: https://youseetoo.github.io/