esphome/esp-web-tools #319: Problem with most current (11.2.0.167) CP210X driver on Windows 10

dkalliv, 2023-01-24:
I was unable to load firmware to some of my ESP32 and ES32-S3 devices (using Edge on Windows 10). Initially this looked random, but after testing about a dozen devices, I noticed that the problem occurred only devices using a Silicon Labs CP210x USB to UART chip. Interestingly enough, I was able to upload firmware to the very same devices using a different computer. After a few tests it seems that CP210X driver version 11.2.0.167 does not play nicely with esp-web-tools or esptool.js (no problems whatsoever using Platformio or arduino IDE). Downgrading the driver to 10.1.9.2677 or even 6.7.4.261 resolves the problem. Hopefully this will save somebody else some time…

Oh, here is one observation that might help to narrow down the cause of the “incompatibility”: if some of the affected devices would eventually go in download mode, this would always happen after (give or take) exactly five minutes...

TD-er, 2023-01-24:
Does Windows have marked this device to enable power save mode? (not sure how it is called, something like "reduce power" or something like that)

dkalliv, 2023-01-25:
I guess you mean "Allow the computer to turn off this device to save power". Yes this is checked (with all the drivers), but it doesn't make any difference. I've tried driver version 11.2.0.167 with this setting unchecked, but still no luck. It looks like esp-web-tools can communicate with the device (I see activity on the tx/rx leds and the device resets) but it can not put the device in download mode. I wonder why this works with other applications like PlatformIO and arduino IDE..?

TD-er, 2023-01-25:
This toggling is done using the DTR/RTS pins.
Maybe those are no longer made accessible in this driver, or are not changing quickly enough?

dkalliv, 2023-01-25:
Unfortunately I do not have a logic analyzer to check the timing of the DTR/RTS pin toggling. However, since PlatformIO and the Arduino IDE are able to put the devices in download mode, the driver clearly supports toggling the DTR/RTS pins. Do you have a logic analyzer?

TD-er, 2023-01-25:
I do have one, but since I'm highly dependent on this driver, I'm not sure I would like to mess with it on my dev PC.
Perhaps if I clean up my desk, I could use one of the laptops here... Maybe a 14" would fit so I can just stack the mess not that high for it to become a liability ;)

dkalliv, 2023-01-25:
This is why I have a 12.5" Notebook for these tasks (I guess the mess on my desk beats your's) ;-)
Let me know what you see on the traces ...

balloob, 2023-01-26:
@dkalliv it would be best to report this upstream as this sounds like an esptool-js issue.

dkalliv, 2023-01-28:
I will check out how esptool-js behaves with this driver and report this upstreams...

dkalliv, 2023-02-04:
I can confirm, esptool-js is getting a timeout when using the latest drivers. I reportet this to esptool-js...

closed 2023-02-04 by dkalliv