esphome/esp-web-tools #320: ESP32-S3 not supported

lecrams, 2023-01-25:
Despite what the documentation says, espweb comes with the message ESP32-S3 not supported.

[image]

I am trying to flash a lilygo T-camera-S3. The device is in a bootloop with the message invalid header 0xffffff.
The bootloop start with this message:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049b21
invalid header: 0xffffffff
invalid header: 0xffffffff

Any ideas how to solve this?

balloob, 2023-01-25:
Can you keep the browser console open while you start the installation and screenshot the results ?

lecrams, 2023-01-26:
On device Dashboard I select install esphome. Then confirm installation (v2022.3.0)
After a few seconds I get:
[image]

Back and then select logs & console. 
[esp-web-tools-logs.txt](https://github.com/esphome/esp-web-tools/files/10507245/esp-web-tools-logs.txt)

This gives an empty console window.
Then reset device gives :
[image]

Hope this is sufficient logging?

balloob, 2023-01-29:
I meant the console that you see when you click anywhere on the page -> inspect element. Then the browser dev tools open and it has a console tab

balloob, 2023-01-29:
oh actually, I see what's up here. The ESP Web Tools demo on the website does not have a build for the ESP32 S-3. It's supported if you deploy your own version and provide a build in the manifest.

closed 2023-01-29 by balloob

balloob, 2023-01-31:
We've updated our demo to now include S3 support 👍

lecrams, 2023-01-31:
Super! saves me some work. Thanks