esphome/esp-web-tools #332: "Failed to fetch" with path setted to remote host with port

hpsaturn, 2023-02-13:
## Overview

Hello again. I was updating my espweb tools version, and I can see that in the documentation mention the possibility of put remote hosts in the manifest. But I tried it with:

```json
{
  "name": "CanAirIO TTGO_T7",
  "version": "rev931",
  "new_install_skip_erase": false,
  "builds": [
    {
      "chipFamily": "ESP32",
      "parts": [
        { "path": "http://influxdb.canair.io:8080/releases/webi/canairio_dev_rev931.bin", "offset": 0 }
      ]
    }
  ]
}
```

and I have this error, "Failed to fetch". With relative path (to the same static web that contains the manifest), it works fine.

Maybe the issue is regarding the port in the url?

Thanks.

## Version

```
<script type="module" src="https://unpkg.com/esp-web-tools@9.2.0/dist/web/install-button.js?module"></script>
```

balloob, 2023-02-13:
Your remote host needs to support CORS and HTTPS. For more info, see the links in our documentation.

closed 2023-02-13 by balloob

hpsaturn, 2023-02-15:
Sorry, I forget this paragraph of your documentation. 
Thanks again, is working now:
[image]