esphome/esp-web-tools #335: [Documentation] Improv WiFi protocol example from scratch for Arduino apps

hpsaturn, 2023-02-15:
## Overview

The ESPWeb Tools it's a great and wonderful  tool. Right now is working in our open source project, and it help a lot of people, but it could be more. Is possible maybe describe or add a basic sketch for Arduino code without ESPHome firmware, to try to have WiFi provisioning and other variables using Improv protocol? Also because right now for example the option in the manifest `"new_install_prompt_erase": True` doesn't works on any firmware that don't have Improv protocol service.

I found the core [implementation of ESPHome](https://github.com/esphome/esphome/blob/dev/esphome/components/improv_serial/improv_serial_component.cpp), the two components, but it needs some components of ESPHome. I think that is possible a basic code for "Arduino developers"

Thanks in advance.

jnthas, 2023-03-22:
Hey! I wrote an example, it might be useful for you. https://github.com/jnthas/improv-wifi-demo

balloob, 2023-03-22:
@jnthas do you want to open a PR to add a link to your project in the docs?

jnthas, 2023-03-22:
Sure! I do it tomorrow :)

hpsaturn, 2023-03-22:
> Hey! I wrote an example, it might be useful for you. https://github.com/jnthas/improv-wifi-demo

Thanks, I will test it tomorrow. Thanks in advance :D

closed 2023-03-23 by balloob