esphome/esp-web-tools #336: M5StickC works with 9.0 but fails with current codebash in read_flash_id

davepl, 2023-02-17:
I had esp-web-tools working against the published UNPKLG 9.0b, but it fails with a timeout against the current code if I built the scripts myself.  It can set the WiFi and do everything BUT flash.

Debugging it, I can see that the following code fails which triggers the timeout message:

```
    async flash_id() {
        this.debug("flash_id");
        const e = await this.read_flash_id();
        this.info("Manufacturer: " + (255 & e).toString(16));
        const t = e >> 16 & 255;
        this.info("Device: " + (e >> 8 & 255).toString(16) + t.toString(16)),
        this.info("Detected flash size: " + this.DETECTED_FLASH_SIZES[t])
    }
```

However, if I run esptool manually for flash_id, works:

```
% esptool.py --port /dev/cu.usbserial-F152E5101E flash_id   
esptool.py v4.4
Serial port /dev/cu.usbserial-F152E5101E
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:9f:6d:09:12:b4
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
```

So something has changed between the two versions, perhaps I'm not supporting a command I need to be, but given it works from the command line I'm at a bit of a loss as to why read_flash_id fails from esp-web-tools.

Not sure if this is a regression or my error, but would sure love to get it working!  

Any ideas?

balloob, 2023-02-17:
The timeout issue was solved upstream in esptool-js. We're just waiting for a release.

davepl, 2023-02-17:
Thanks!  Any rough sense on the ETA (days, weeks, etc)?

balloob, 2023-02-17:
Nope. I had asked them for a release [here](https://github.com/espressif/esptool-js/pull/83#issuecomment-1416295465) and they responded positive. However, nothing has happened yet.

davepl, 2023-02-18:
Thanks for the info!  Is there a "last known good" version that I can grab from the Releases page that works?