esphome/esp-web-tools #338: wemos s2 mini installation fails

Talitsch, 2023-02-22:
installation of wemos s2 mini https://www.wemos.cc/en/latest/s2/s2_mini.html fails with "Fails to initialize". 
Chrome log for installation:

[image]

Talitsch, 2023-02-22:
the only online esp browser tool that does not disconnect while uploading something to the board is this one: https://adafruit.github.io/Adafruit_WebSerial_ESPTool/

balloob, 2023-02-24:
waiting for a release of esptool-js upstream.

closed 2023-02-28 by balloob

dkebler, 2024-01-06:
still an issue.  has esptool-js been updated to fix this?

dkebler, 2024-01-06:
I think the issue is that the webtool by default does a `--before hard-reset`.  That takes the s2 out of programming mode and thus the error.

When I use esptool.py directly I use the below putting the esp32s2 in programmer mode before runining esptool.

```
esptool.py  \
         --port $port         \
         --before no_reset               \
         --after hard_reset  \
         --baud 115200                \
         --chip esp32s2                \
         write_flash 0 $bin   
```

spuder, 2024-11-03:
I'm seeing this exact behavior with Wemos D1 Mini S2 and the latest sep-web-tools (10)

`https://unpkg.com/esp-web-tools@10/dist/web/install-button.js?module`

I have better luck with the adafruit web upload tool, but it has some other limitations that make it not viable. 

https://adafruit.github.io/Adafruit_WebSerial_ESPTool/