esphome/esp-web-tools #366: Failed to download manifest

Lasakro, 2023-05-04:
I've seen 2 resolutions for this error. Both state: Resolved, add "/" in path folder name and work fine.

1) I goto: https://esphome.github.io/esp-web-tools/ and select connect.
2) I choose the CP2102 for my esp32 in flash mode and again select Connect.

This is where I receive Error Failed to download manifest. What am I missing?

jnthas, 2023-05-06:
You are not missing anything. It was a misleading update few weeks ago that cause this bug. 
Just wait the PR be merged and it will back to normal
https://github.com/esphome/esp-web-tools/pull/367

anubis-code, 2023-05-09:
Hi when will this be getting merged as it is still not working

drumstick77, 2023-05-12:
Hi this is still broken
any update please?

jnthas, 2023-05-12:
@balloob @jesserockz could you take a look into this, please?

drumstick77, 2023-05-12:
Thanks @jnthas , I found that by going to https://web.esphome.io/, it seems to be working okay. Just not on the live demo page. Hope that helps

balloob, 2023-05-20:
Sorry for late response. Merged the fix.

closed 2023-05-20 by balloob