esphome/esp-web-tools #374: ESP32-C3 works well S2 and S3 not.

sblantipodi, 2023-05-16:


TD-er, 2023-05-16:
I am flashing C3/S2 and S3 just fine using [my setup](https://td-er.nl/ESPEasy/latest/)
As you can see in the source of my page, I am using 8.0.6
So maybe you can try using my setup, and/or switching to that version to test yourself.

sblantipodi, 2023-05-16:
thanks again for the answer.
yes I tried that but same result.

The strange thing is that as soon as I connect the ESP it shows as COM20
[image]

if I press the IO0 button, then the RST one, I hear the ding dong sound in windows, esp web tools see the device as COM19.
[image]
this thing happens on platformio too. it starts with COM20 but then it flashes on COM19.

I think that ESP Web Tools struggles in this part, S2 is the only device that does this...

This is the PlatformIO output.
```
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM20
Forcing reset using 1200bps open/close on port COM20
Waiting for the new upload port...
Uploading .pio\build\glowwormluciferinfull_esp32s2\firmware.bin
esptool.py v4.5.1
Serial port COM19
Connecting...
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:d7:d8:1a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000effff...
Compressed 13600 bytes to 9549...
Writing at 0x00001000... (20 %)
Writing at 0x00001c2d... (40 %)
Writing at 0x00002799... (60 %)
Writing at 0x000031a7... (80 %)
Writing at 0x00003d2c... (100 %)
Wrote 13600 bytes (9549 compressed) at 0x00001000 in 0.2 seconds (effective 570.7 kbit/s)...
```

TD-er, 2023-05-16:
The Wemos S2 mini I have here (USB-C, no separate USB to serial chip) is a bit tricky to flash via the web flasher.
I really need to press the "0" button, keep it pressed, press "RST", release "RST" and then reload the web flasher page, then select the flash procedure, select com port, etc. and when it is showing the progress meter with at least a few % I can release the "0" button.

sblantipodi, 2023-05-17:
@TD-er if I press the button until the end as per your suggestion it works, thank you very much,
this is much appreciated.

I think that the issue can be closed but I have a last answer before closing it.

Do you think that esp web tools can by-pass these problems and flash this devices smoothly as PlatformIO does?

ssieb, 2023-05-17:
It would be difficult.  platformio is switching serial devices half-way through.  The browser web serial is going to have a problem with that.

sblantipodi, 2023-05-17:
ok, closing the issue then, thanks for the help TD-er and ssieb.

sblantipodi, 2023-05-22:
I reopen this issue because the web installer is seriously compromised by this problem,
the entire purpose of the web installer is to make the installation process user friendly and it is not with ESP32-S2 and S3.

I think that this needs some attention...

sblantipodi, 2023-05-26:
@TD-er I'm sorry if I quote you, I downgraded ESP Web Tools from 9.2.1 to 8.0.6 and now it works since it starts by erasing the device.

Why newer version of ESP Web Tools works worse than the older one? :)

TD-er, 2023-05-26:
> Why newer version of ESP Web Tools works worse than the older one? :)

No idea :)

Jason2866, 2023-06-08:
v8.0.6 is not using espressif esptool.js under the hood. All later use espressif esptool.js. It is a (until now not found) bug in espressif esptool.js

TomasHubelbauer, 2023-08-10:
@TD-er can you please walk me through the process you described for successfully flashing the Wemos S2 Mini in more detail? I tried to follow it:

- Start off at https://esphome.github.io/esp-web-tools/ with an unpaired and disconnected Wemos S2 Mini
- Press 0 and continue holding it until the step that says otherwise
- Connect the board via USB C to the computer
- Press and release RST
- Click Connect on the page
- Select the unpaired board and click Connect in Chrome
- Wait for the page to show and hide the Connecting dialog
- Refresh the tab
- Click Connect on the page
- Select the now paired board and click Connect in Chrome
- Wait for the page to show and hide the Connecting dialog
- Click the Install ESPHome button
- Click Install
- Continue holding the 0 button on the board
- Wait as the preparing modal is in progress

I believe I am following your steps correctly but in my case I always end up with an error at the end of this procedure. Did I misunderstand your steps? Or maybe the procedure is flaky - does it work for you every time with these steps?

sblantipodi, 2023-08-10:
what works for me is:
- connect USB C
- set the device in "flash mode"  by pressing 0 + RST, in windows you'll hear a ding sound like if a USB device has been detached
- release the RST button, you'll hear a ding again like a USB device has been atteched, don't leave 0 button, keep it pressed.
- start flashing using web installer with the 0 button pressed
- when it starts writing the firmware that you see the loading progress 5% for example, you can release the 0 button...

it's a very bad experience right now

TomasHubelbauer, 2023-08-10:
Thanks for the steps! Unfortunately it seems that in my board's case it just doesn't want to work. I get an error the moment it should start flashing even though I believe I am following the steps correctly. I am new to ESP32 so I am not sure this is the key but I think I will give it a shot with a non-S2 board to see if it works better for me.

TomasHubelbauer, 2023-08-10:
@sblantipodi Much better! I was able to flash the board with no special tricks, just by following the instructions for S2:
[image]
The flashing procedure went through successfully. I was not prompted to set a wi-fi password after the flash so I reset the board and clicked the Install button again. At this point I did see an option to set the wi-fi and I put in my SSID and password, but I got an error message that said "Disconnected". My SSID has a space in it which I think might be causing issues. This is unrelated to the flasher I think so probably not relevant?

sblantipodi, 2023-08-10:
@TomasHubelbauer yes that flasher is stil not that stable with S2.
hope to see a better support in the future.

TD-er, 2023-08-10:
I also use the one put together by @Jason2866 
See here my flasher page: https://td-er.nl/ESPEasy/

After the flashing has started, you can let go of the '0' button.

Sometimes you need to power cycle the ESP32-S2 in order to get it to work.

espilioto, 2023-12-05:
Got a Wemos S2 mini and still cannot get it working with esphome or wled.
The adafruit tool worked immediately as did luciferin interestingly.

Could this be a case of the individual projects not updating their esp-web-tools version?

Jason2866, 2023-12-05:
@espilioto The use the esphome web tools, which have espressif esptool.js under the hood.
My Web tools uses a fork of Adafruit Web serial esptool which i have enhanced for supporting C2 and C6. The changes are provided via a PR upstream to Adafruit and are merged.
So the the Tasmota Web tool is totally different to the esphome Web tool.

espilioto, 2023-12-05:
Oh ok, that makes sense. 
Thanks for the reply. 

(I tried esphome and wled, both failed)

sblantipodi, 2023-12-10:
@espilioto just for reference, 
Luciferin is now using the web tools from Jason2866 under the hood.

hpsaturn, 2023-12-31:
Hi everyone,

I have the same problem with S3 variants in my implementation. I'm using the last version, and it works fine for esp32 and esp32c3. I did also a test with the ESPHome installer and it works fine over my testing board. I think that maybe I have some issue with the merge binary generation? Maybe I have a mistake in this line?

```bash
/usr/bin/python3 ${PIO_HOME}/packages/tool-esptoolpy/esptool.py --chip esp32s3 merge_bin -o  $FIRMDIR/${NAME}_${1}_rev${SRC_REV}_merged.bin --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0000 $OUTDIR/$1/bootloader.bin 0x8000 $OUTDIR/$1/partitions.bin 0xe000 ${PIO_HOME}/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 0x10000 $OUTDIR/$1/firmware.bin
```

[image]
[image]
[image]

**My implementation is here:**
https://canair.io/installer

And I'm using the ESP32S3 boards: TTGO T7 S3 and ESP32S3 Freenove (ESP32S3 CAM)

Thanks in advance.

hpsaturn, 2023-12-31:
Sorry, I found my mistake. In my CI script that write the manifest files, I forget add a rule for the ESP32-S3 family. That was the problem. Right now is working my new two variants for ESP32S3 🎉. Thanks.

dkebler, 2024-01-18:
In a nutshell tasmota/weld installed and working but not esphome for reasons explained here.  

https://community.home-assistant.io/t/cant-install-esphome-on-s2-mini/621559/6?u=dkebler

sblantipodi, 2024-02-13:
it seems that the latest release 10.0.1 improved this problem.
I don't know if it was improved previously but 10.0.1 works better now on ESP32-C3, S2, S3.

ESPs can be flashed without problems by setting them in flash mode, improv-wifi does not work always and you need to retry and retry before getting the "change wifi" option.

danch99, 2024-03-16:
I've successfully setup my s2 mini following this : [https://www.reddit.com/r/Esphome/comments/11a08za/comment/j9pes7c/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button](https://www.reddit.com/r/Esphome/comments/11a08za/comment/j9pes7c/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

Basically using Adafruit ESPTool to upload the first firmware. And then I ca see it in ESPhome and update OTA